aredn/aredn #1776: USB port intermittent power or not powering  20241217-68837dbb hap ac2 and hap aclite

kp4djt, 2024-12-22:
### Describe the bug
Noticed that GPS data was not being captured after moving to nightly 20241217-68837dbb on the hap ac lite and intermittently on the hap ac2. I telneted to both units and first looked to see if the ttyACM0 was present. One the hap ac2 it would initially show up after a reboot then disappear. One the hap ac lite it never showed up. 
On hAP ac2 the USB power shows green but appears no power being applied
On hAP ac lite, USB power shows red even though the setting is for USB power. 

- [ ] Does this issue still occur on the latest version? (**Required**)
Running latest nightly  20241217-68837dbb it shows up, I downgraded to 3.24.10.0 and observed that the issue goes away 

### Expected behavior
Port should have power applied when the USB power is set on. I have tested with gps radios, of course when powered
I see a data stream off of ttyACM0.

### Support data
Tested with 3.24.0.0 and both nodes power up the USB port as indicated by the softswitch. Also using a gps radio
I am able to demonstrate that there is power to the radio under 3.24.10.0
Testing with nightly  20241217-68837dbb, hap ac lite shows USB power off regardless of setting

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
I have only been able to test this with these two devices.

aanon4, 2024-12-23:
Please provide support data for both nodes while in their failure state. If you can capture it while working also that'd be nice.

aanon4, 2024-12-23:
Also - regarding the hap ac2 - there are no software controls for the power - it just does what it does and we dont get any say in the matter.

kp4djt, 2024-12-23:
OK, probably do it when I get back home this week. Yesterday I was killing
time at the hotel.

I had loaded 3.25.10.0 back into both and they were working so did not try
to go back. Was also fightinig
with a meshtastic device too.

Keep up the good work.

aanon4, 2025-01-28:
Closing as there was no feedback in the last month.

closed 2025-01-28 by aanon4