aredn/aredn #1797: Add Qemu Guest Agent support using qemu-ga package

cameronzucker, 2025-01-02:
**Is your feature request related to a problem? Please describe.**
The x86 VM can't be properly managed in Proxmox without running qemu guest agent in the VM itself. OpenWRT does have a package for this ([here](https://openwrt.org/packages/pkgdata/qemu-ga)), but it's not included in the default AREDN package library and resists every attempt I've made at manually installing it. Configuring VMs with qemu guest agent allows them to accept commands like bulk shutdown, restart, etc.

**Describe the solution you'd like**
Please add qemu-ga to the default AREDN package list so it can be installed on the current VM build.

**Describe alternatives you've considered**
There are no alternatives to running Qemu Guest Agent that I'm aware of.

**Additional context**
This is a fairly straightforward request which shouldn't require additional context.

aanon4, 2025-01-02:
I've made it available as a loadable module.

cameronzucker, 2025-01-03:
Is this something that I'll need to be on the next nightly firmware build to see? Checked the package list earlier when I saw your update and couldn't find anything that looked like qemu-ga.

aanon4, 2025-01-03:
We build once a day, usually available at noon UTC. It wont be available until then but I'd appreciate verification once that happens - thanks.

cameronzucker, 2025-01-03:
Installed and working. Responds to Proxmox commands as expected.

closed 2025-01-03 by cameronzucker