aredn/aredn #1805: Move "Tools" Menu Icon up 1/2 to 1" on Left Menu Bar to avoid being covered by static Windows Toolbar

k1ky, 2025-01-06:
**Is your feature request related to a problem? Please describe.**
The "Tools" Icon can be covered up by a Static or "Stuck" Windows Menu Bar.
**Describe the solution you'd like**
Fix: Move the Tools Icon selection up 1/2 to 1" from the bottom in order to avoid being covered up by the Windows Menu Bar.
**Describe alternatives you've considered**
None
**Additional context**
Example of issue - there is a Tools Icon that is Covered up by the Menu Bar:
[image]

k1ky, 2025-01-06:
Anywhere even with the Firmware Version line would be acceptable.

aanon4, 2025-01-06:
I'm not sure I understand. The icon is at the bottom of the browser page. If it's under the toolbar doesn't this mean your browser is also under the toolbar all along the bottom?

k1ky, 2025-01-06:
That is correct, my toolbar is apparently covering up the bottom 1/2" of the browser screen in Full Screen (Maximized) mode.  Moving the selectable Tools Icon makes it accessible in this situation.

aanon4, 2025-01-06:
Doesnt it make more sense to not put your browser under your toolbar? Dont you have this problem with everything you browse being cut off?

k1ky, 2025-01-06:
That would be a good solution, it just happens to be the way mine is set up - and if it was, then how would I access the tool bar?  Usually it autohides, but there are occasions when the toolbar is stuck on (possible Windows bug?).  At any rate, this the first time I have observed this issue and was providing a solution/suggestion in case other folks encountered this problem.

closed 2025-01-07 by aanon4