aredn/aredn #1812: hAP ac lite horrible reliability on NB 20250109 or 20250110

k6ccc, 2025-01-10:
### Describe the bug
MikroTik RouterBOARD 952Ui-5ac2nD (hAP ac lite) unable to maintain either DtD nor WireGuard links for more than a short time (maybe an hour - give or take a lot)

- [ ] Does this issue still occur on the latest version? (**Required**)
Initially updated from 20241231 to 20250109 and had horrible reliability, downgraded back to 20241231 and working perfectly.  Tried 20250109 again with the same result.  Downgraded again to 20241231 and working perfectly.  This morning tried 20250110 with similar results to 20250109.

### Expected behavior
A DtD and WG link should be stable.

### Support data
Unable to download support data.  AREDN logo spins for about a minute and then web browser tab spins for another couple minutes.  The best I can get is a partial download - i.e. Unconfirmed 286531.crdownload file

### Screenshots
nothing useful.

### Additional context
This hAP is connected to a HP managed switch along with two Rocket M5 nodes.  The switch is connecting VLAN 2 for the DtD connection and VLAN 1 for the WAN connections.  All three nodes have WG tunnels.  The tunnels and DtD between the Rockets work perfectly.  I am accessing these nodes via the WAN IP (because it is easier that way from this work computer).  This node does not have any RF connections.
At times I could not even access the web page.  Had to SSH into the node and reboot it.  Then the web page would work.

k6ccc, 2025-01-10:
I have reverted it back to 20241231, but should be able to try 20250110 again.
I was able to SSH in.  How do I get Support Data via SSH?

aanon4, 2025-01-10:
```
/usr/local/bin/supportdata
```
Will generate the support data. Use `scp` (or whatever the Windows equivalent is) from your host machine to copy it off the node.

k6ccc, 2025-01-10:
Back on 20250110.  This is from right after the reboot.
[image]
[supportdata-K6CCC-hAP-Glendale-2025-1-10-11-0.tar.gz](https://github.com/user-attachments/files/18381266/supportdata-K6CCC-hAP-Glendale-2025-1-10-11-0.tar.gz)

aanon4, 2025-01-10:
Thanks. If you can follow this up once its behaving badly that'd be great.

aanon4, 2025-01-10:
It's running out of memory.
Now we have to work out why. Could you start be stopping `K6CCC-HAP-GLENDALE-172-31-231-36` which is continually trying and failing to connect its tunnel. Want to eliminate that as a possible memory leak.

k6ccc, 2025-01-10:
BTW, even without being on the SoCal network, you can see status of that node by pointing to this URL.  It updates every 5 minutes and this node is line 1.
http://k6ccc.org/aredn_test_results/AREDN_Ping-now.html

aanon4, 2025-01-10:
Its okay if the tunnel is on, I just dont want it failing to connect over and over and over. It also ends up being what the log files mostly report rather than something more useful.

k6ccc, 2025-01-11:
After disabling the legacy tunnel a couple hours ago, the hAP has been solid.

k6ccc, 2025-01-11:
I spoke too soon.  All links failed again.

aanon4, 2025-01-11:
Sorry .... I still need the support data after it failed. Probably a bit late now as it's been hours since the failure and as close to when it happened would be idea.

k6ccc, 2025-01-12:
Unfortunately I'm not at work and with it not working, I have no way to get the support data until Monday when I get back to work.

aanon4, 2025-01-12:
No rush. When you do get back to it, please start from whatever the current nighty is at that time. Thanks.

k6ccc, 2025-01-12:
Will do.

k6ccc, 2025-01-13:
OK, here you go.  Came into work this morning and could locally connect to the this hAP via the WAN port.  It was not connected to any other nodes.  First I downloaded the support data (1-13-6-27).  Then initiated a firmware update from 20250110 to 20250113 using the Upload Firmware.  The update never completed - after selecting the file and clicking "Fetch and Update", the bar just kept bouncing back and forth.  Let it do so for a bit over 20 minutes.  Attempted to open another web session and it would not connect.  Connected via SSH and commanded a reboot.  After the reboot, I was able to connect and it did establish the DtD connections.  Downloaded support data a second time (1-13-6-59), and then did the update to 20250113.  This time it worked fine.  After the node updated, as soon as it was able to get time, I downloaded support data for a 3rd time (1-13-7-7).  All three of those file are attached.
Now we wait to see if / when it fails....
[supportdata-K6CCC-hAP-Glendale-2025-1-13-6-27.tar.gz](https://github.com/user-attachments/files/18398573/supportdata-K6CCC-hAP-Glendale-2025-1-13-6-27.tar.gz)
[supportdata-K6CCC-hAP-Glendale-2025-1-13-7-7.tar.gz](https://github.com/user-attachments/files/18398575/supportdata-K6CCC-hAP-Glendale-2025-1-13-7-7.tar.gz)
[supportdata-K6CCC-hAP-Glendale-2025-1-13-6-59.tar.gz](https://github.com/user-attachments/files/18398576/supportdata-K6CCC-hAP-Glendale-2025-1-13-6-59.tar.gz)

k6ccc, 2025-01-14:
The node started losing connectivity about 3 something this morning and is intermittent.  Here is a screen capture and support data from the first failure I caught after getting to this location.
[supportdata-K6CCC-hAP-Glendale-2025-1-14-6-50.tar.gz](https://github.com/user-attachments/files/18411888/supportdata-K6CCC-hAP-Glendale-2025-1-14-6-50.tar.gz)
[image]

aanon4, 2025-01-14:
Thanks. Still running out of memory.
Is it possible for you to disable the LAN Hotspot and still have access to the node for debugging? That's the process which keeps running out of memory, or at least causing the system to kill random things to get more memory.

aanon4, 2025-01-14:
Dont worry about updating to nightly today (you can, but its a minor config change to other sorts of devices)

k6ccc, 2025-01-14:
Done.  Neither 2.4 nor 5 GHz radios are needed on this at all.  Both are now turned off.
We'll see what happens.

k6ccc, 2025-01-14:
Not that you asked for it, but after turning off both radios, and the required reboot, I downloaded support data as a baseline.  Here that is.
[supportdata-K6CCC-hAP-Glendale-2025-1-14-8-32.tar.gz](https://github.com/user-attachments/files/18413391/supportdata-K6CCC-hAP-Glendale-2025-1-14-8-32.tar.gz)

k6ccc, 2025-01-15:
24 hours and still working...

aanon4, 2025-01-16:
Okay. I'm switching to a different hostapd version (OpenWRT has lots to choose from).
https://github.com/aredn/aredn/pull/1822
If you could update and re-enable your LAN Hotspot that would be lovely.

aanon4, 2025-01-17:
Still working?

k6ccc, 2025-01-17:
Yes.  Uptime just short of 24 hours.

k6ccc, 2025-01-17:
31 hours 28 minutes and working fine.  Although no needed, I am going to enable Mesh on the 2 GHz radio.  However prior to that, I downloaded support data.  Here it is.

[supportdata-K6CCC-hAP-Glendale-2025-1-17-15-16.tar.gz](https://github.com/user-attachments/files/18461618/supportdata-K6CCC-hAP-Glendale-2025-1-17-15-16.tar.gz)

k6ccc, 2025-01-20:
Hmmm, the node went unknown from home for a while this evening and then restored after a while.  Logged into it and downloaded support data.  Here it is:

[supportdata-K6CCC-hAP-Glendale-2025-1-19-20-15.tar.gz](https://github.com/user-attachments/files/18472214/supportdata-K6CCC-hAP-Glendale-2025-1-19-20-15.tar.gz)

aanon4, 2025-01-20:
Something it still leaking as it's still running out of memory. I assume you just ignoring it for the most part?

aanon4, 2025-01-20:
Could I get login access to the node? I'm going to need to catch it just before it runs out of memory.

k6ccc, 2025-01-20:
Yes.  Sent via E-Mail

aanon4, 2025-01-20:
Well I tried to upgrade it so we can work from the current baseline but now its dead :-(

k6ccc, 2025-01-20:
Correct.  The backup legacy tunnel is normally left turned off at my house.  I have enabled it and disabled the WG tunnel.  I will be into the office Tuesday morning so I'll see what I see when I get there.

This node is essentially doing nothing useful, so having it dead / failing / rebooted is not an issue.  I have two Rockets at my office that point to two different mountains.  They are DtD connected and both have a tunnel to my house.  A while back (year or two) I was having issues with those two nodes and to help rule out issues, I added the hAP to the DtD and had the hAP tunnel to the house and disabled the tunnels from the Rockets to the house.  Turned out the issue was a failing network switch that carried the WAN connections to all three nodes, so once I ruled out a AREDN issue, I re-enabled the Rocket tunnels.  That leaves the hAP not really doing anything useful.  Can be useful when I am doing firmware updates as it leaves another reliable path.

BTW, if you are not connected to the SoCal mesh, you can quickly see status of this at: http://k6ccc.org/aredn_test_results/AREDN_Ping-now.html  This page updates every 5 minutes.

k6ccc, 2025-01-20:
Entertaining note.  I was able to remote into the LAN that is used for the WAN connection on the three nodes at work.  I can ping the WAN IP, but from that LAN, and unable to SSH or Web GUI into the hAP.

k6ccc, 2025-01-21:
Got into work this morning and again confirmed I could not access the node via SSH or Web GUI, but it did respond to pings.  Went upstairs and power cycled it.  After that it was working normally.  Updated the firmware to 20250121-b46f49a9 without issue.  About 45 minutes later grabbed this screen capture and a support file.  Note that although this hAP is not showing the links correctly, the two Rockets and the hAPac3 at the other end of the DtD or tunnel connections are showing the connections properly.

[supportdata-K6CCC-hAP-Glendale-2025-1-21-7-24.tar.gz](https://github.com/user-attachments/files/18492977/supportdata-K6CCC-hAP-Glendale-2025-1-21-7-24.tar.gz)

[image]

k6ccc, 2025-01-21:
Well, crapped out again.

[image]

[supportdata-K6CCC-hAP-Glendale-2025-1-21-8-57.tar.gz](https://github.com/user-attachments/files/18494088/supportdata-K6CCC-hAP-Glendale-2025-1-21-8-57.tar.gz)

k6ccc, 2025-01-21:
Also note, I will be out in the field for several hours and unable to do anything with this for a while.

aanon4, 2025-01-21:
No worries. Seems to be down at the moment though.

aanon4, 2025-01-21:
[supportdata-KO6M-QTH2-2025-1-21-16-44.tar.zip](https://github.com/user-attachments/files/18496107/supportdata-KO6M-QTH2-2025-1-21-16-44.tar.zip)
Support data from another user with similar problems.

k6ccc, 2025-01-21:
Yes, it's down.  I am back from Mt. Lukens.  I can still access the web GUI via the WAN port.  I did a support data shortly after it crapped out.  do you want another one?

aanon4, 2025-01-22:
If you haven't already, please disable the wireguard tunnel and switch to your backup legacy tunnel instead. Then reboot so we start from a clean slate with this test.

k6ccc, 2025-01-22:
As seen in my first screen capture this morning, the WG tunnel is not active and the legacy tunnel is active.  That was 37 minutes after updating firmware to 20250121.  A couple hours later it failed again.

aanon4, 2025-01-22:
Weird. When I checked (and just rechecked) that support data with that image, the wireguard tunnel is enabled in the network file. I just checked the latest one and it's not there now. So ... odd.

aanon4, 2025-01-22:
I use to be able to reach the node, but not anymore :-(

k6ccc, 2025-01-22:
I see it's crapped again.

aanon4, 2025-01-22:
Does it seem more unreliable like this than before?
One question and one thing to try:

Question. Tell me about the WAN links at each end of the tunnel?

Try. Let's disable all tunnels. I dont know if this makes it unreachable or not, but hopefully there's another way in.  If not, let's at least try this while you're in the office.

k6ccc, 2025-01-22:
Yes, it definitely worse today.

At the Glendale end, WAN is from a symetrical Spectrum cable link with 50 Mb/s up and down.  It passes through at least one router (Juniper enterprise grade) and a couple switches.  The two Rockets are plugged into the same switch and are rock solid.

At my house, the WAN is from a symetrical gigabit Frontier fiber.  It passes through a Mikrotik RB4011 router and a couple Mikrotik CSS326 switches.  The slowest link between Fiber ONT and the hAPac3 is gigabit.

No problem killing the tunnel completely.  As long as the DtD connections are working, the hAP can be reached from here via the Rocket tunnels.  I will attempt to get through to kill the tunnel completely.

k6ccc, 2025-01-22:
I can't get into it well enough to command anything via the GUI and could not SSH into it at all.  Does respond to a ping however.
I'll see what I can do locally in the morning.

k6ccc, 2025-01-22:
This morning I was able to access the hAP locally via the WAN port.  It showed no connections.  Downloaded support date.  Then updated to 20250122.  A few minutes later all connections were up - including the legacy tunnel to home (had not been able to disable it last night remotely).  Downloaded support data again.  Then turned off the tunnel to home, waited a while and downloaded support data a third time.

[supportdata-K6CCC-hAP-Glendale-2025-1-22-6-7.tar.gz](https://github.com/user-attachments/files/18507187/supportdata-K6CCC-hAP-Glendale-2025-1-22-6-7.tar.gz)

[supportdata-K6CCC-hAP-Glendale-2025-1-22-6-25.tar.gz](https://github.com/user-attachments/files/18507189/supportdata-K6CCC-hAP-Glendale-2025-1-22-6-25.tar.gz)

[supportdata-K6CCC-hAP-Glendale-2025-1-22-6-50.tar.gz](https://github.com/user-attachments/files/18507190/supportdata-K6CCC-hAP-Glendale-2025-1-22-6-50.tar.gz)

k6ccc, 2025-01-22:
And just a few minutes later, the DtD connections are down.  

[supportdata-K6CCC-hAP-Glendale-2025-1-22-6-57.tar.gz](https://github.com/user-attachments/files/18507333/supportdata-K6CCC-hAP-Glendale-2025-1-22-6-57.tar.gz)

aanon4, 2025-01-22:
[aredn-kn6plv-main-dc1f8d1e-ath79-mikrotik-mikrotik_routerboard-952ui-5ac2nd-squashfs-sysupgrade.bin.zip](https://github.com/user-attachments/files/18509210/aredn-kn6plv-main-dc1f8d1e-ath79-mikrotik-mikrotik_routerboard-952ui-5ac2nd-squashfs-sysupgrade.bin.zip)
Well we're at the custom kernel testing part of this. Could you uncompressed and install this kernel for testing - thanks.

k6ccc, 2025-01-22:
KN6PLV-main-dc1f8d1e loaded.  Support data as a baseline.

[supportdata-K6CCC-hAP-Glendale-2025-1-22-7-39.tar.gz](https://github.com/user-attachments/files/18509369/supportdata-K6CCC-hAP-Glendale-2025-1-22-7-39.tar.gz)

k6ccc, 2025-01-22:
One hour and counting.  Good so far...

aanon4, 2025-01-22:
Eh ... I'm not getting my hopes up anymore ...

aanon4, 2025-01-22:
Still failing and running out of memory.
Next thing we do is unplug the thing from the mesh, reboot it, and just let it sit there on its own to see if, with nothing mesh happening, it can not run out of memory.

k6ccc, 2025-01-22:
I see that too.  
As far as disconnecting, it currently has 5 GHz WiFi hotspot (not being used), a 2 GHz Mesh node (not doing anything), and a DtD to the two co-located Rockets.  I can disconnect any or all of that.  What do you want?

aanon4, 2025-01-22:
Lets turn off everything but whatever you use to connect to it locally (wired LAN I'm assuming). Might as well got with the very basics.

k6ccc, 2025-01-22:
My wired connected is to the WAN port, but I will kill everything else.

k6ccc, 2025-01-22:
OK, it's now very lonely.  Here's a baseline support data.

[supportdata-K6CCC-hAP-Glendale-2025-1-22-13-32.tar.gz](https://github.com/user-attachments/files/18511858/supportdata-K6CCC-hAP-Glendale-2025-1-22-13-32.tar.gz)

k6ccc, 2025-01-22:
Obviously you can't access it anymore.  I can do so when I'm at work and somewhat when I'm at home via a VPN.

aanon4, 2025-01-22:
Let's just ignore it for 24 hours and then check its status tomorrow if you can.

k6ccc, 2025-01-22:
Can do.

aanon4, 2025-01-23:
Morining. When you check on the node next, grab the support data (I know you will anyway).
After that, upgrade it to the latest nightly and put it back on the network with DtD links but, not yet, any tunnels.

[14 more replies]

closed 2025-01-29 by aanon4