aredn/aredn #1815: Unidentified Hardware causing crash uhttpd on firstboot

Innismir, 2025-01-11:
### Describe the bug
Upon install of AREDN on hardware on a platform that is unable to be identified by getBoard(), the web app crashes upon FirstBoot. This occurred to me when I was installing AREDN on a bare iron x64 system

- [ ] Does this issue still occur on the latest version? 
Present in nightly

### Expected behavior
uhttpd starts normally on firstboot

### Additional context
 My quick fix was eddting getBoard and adding a
`        else {
            boardJson.model.id = "vmware";
            boardJson.model.name = "VMware";
        }`

Which causes the script to just return "VMware" if it can't identify the system. This has successfully worked around the glitch but is a duct tape solution.

aanon4, 2025-01-11:
Please add the support data.

Innismir, 2025-01-11:
[supportdata-N1WBV-DARTMADUMR1-2025-1-11-22-37.tar.gz](https://github.com/user-attachments/files/18387517/supportdata-N1WBV-DARTMADUMR1-2025-1-11-22-37.tar.gz)

aanon4, 2025-01-12:
Hmm ... it identifies itself as "WYSE ZQ Class" which seems very specific. I'd hoped for something a little more generic.

aanon4, 2025-01-12:
https://github.com/aredn/aredn/pull/1816

aanon4, 2025-01-18:
Feedback?

aanon4, 2025-02-04:
Please verify this is fixed.

closed 2025-02-05 by aanon4