aredn/aredn #1820: Add ability to make a node an NTP server on the LAN side (envisioned for when there is no internet NTP access, but with GPS date/time.)

convolved-KY4G, 2025-01-14:
**Is your feature request related to a problem? Please describe.**
Kindof. When deploying go-kits for our ARES SET exercises, we also make use of equipment such as VOIP phones hooked up to our AREDN network that are not connected to the internet.  These phones typically display some random time on 1/1/1970.  This has occurred in multiple AREDN demonstration activities as well and people even point it out.  Now that [#1475 ](https://github.com/aredn/aredn/issues/1475) has been implemented, there is an easy way to fix this.

**Describe the solution you'd like**
Add a toggle switch under the advanced options portion of the "Time" page to allow user to enable if the node is to be used as an NTP server on the LAN side.  When the switch is "on", then the ntp service in busybox can be restarted, but will need to include the "-l" flag in order for it to act as an NTP server.

**Describe alternatives you've considered**
Currently, any changes to the ntp configuration get overwritten from the image of the config file stored in the ROM section of the firmware.  That could be changed by making the NTP portion of the config file persistent for this change, but that will require writing scripts to accomplish that and would only be able to be changed by ssh'ing into the node and using the appropriate commands or editing the file manually.  I also could keep going in after every update and just keep adding a -l to the list of arguments when the ntp service gets called in busybox, but I know of others who would like to solve this same issue.  This could also be done with another  device on the LAN that serves as an NTP server, but this seems like a lot of trouble when the node can perform this simple service itself as an option.

**Additional context**
So, I have tested this by hardcoding the "-l" flag into one of the scripts.  After I set up DHCP option 42 to point to the node, the IP phone picked up the correct time and date in short order and displays it correctly.  I should add that I was using the [whereandwhen](https://github.com/kn6plv/WhereAndWhen) package with a GPS antenna to set the time and date on the node itself.   For the purposes of this enhancement, GPS time is sufficient.

aanon4, 2025-01-29:
https://github.com/aredn/aredn/pull/1865

convolved-KY4G, 2025-01-30:
I tested the nightly build 20250129-d446fe15 (r28417-daef29c75d) on an AR300M16 with no external connectivity except for the GPS antenna. The VOIP phone on the LAN picked up the date and time from the node ntp server as expected. Thanks again for implementing this!

closed 2025-01-30 by convolved-KY4G