aredn/aredn #1833: NB 20250118 Impacts Local Services and Local Devices Displays

K5RA, 2025-01-20:
System here is a hAP ac lite (HAP-FB) DtD to a hAP ac2 (HAP2-FB) DtD to a Rocket XW (RK2C).  All three nodes were running stable release 3.24.10.0.  HAP-FB had an advertisee Dial-by-IP for a VoIP phone.  HAP-AC has a MeshChat package (2.12.1_api) installed, but it was not advertised.  Service name is NEW-NTXChat.  HAP2-FB has MeshChat package (2.12.1_all) installed and advertised.  Node Status screenshot and HAP-FB Support Data attached (2025-01-16).  

When I installed NB 20250118 in HAP-FB, HAP-FB’s Local Service announcement for the VoIP phone was deleted, and a Local Service announcement for Meshchat-1879 was posted.  Also, a Link icon was appended to each displayed Local Device (phone and R-Pi).   Node Status screenshot and HAP-FB Support Data attached (2025-01-18).  

The effects were the result of upgrading 3.24.10.0 to NB 20250118 in a hAP ac lite.

I expect the Local Devices and Local Services displays to be unchanged to be unchanged.  

Before and after screenshots and support data provided

[image]

[supportdata-K5RA-HAP-FB-2025-1-16-13-4.tar.gz](https://github.com/user-attachments/files/18482989/supportdata-K5RA-HAP-FB-2025-1-16-13-4.tar.gz)

[image]

[supportdata-K5RA-HAP-FB-2025-1-18-12-56.tar.gz](https://github.com/user-attachments/files/18482993/supportdata-K5RA-HAP-FB-2025-1-18-12-56.tar.gz)

aanon4, 2025-01-21:
Background: We're migrating some old configuration data into a new format.
This node looks to have been configure in NAT mode and non-NAT mode at different points. Previously (and for reasons lost to time) we kept the services information complete separate for these two modes. It looks like meshchat had been configure when in NAT mode, and the "Dial by ..." text in non-NAT mode.  When we migrated the mode because both were valid we just picked one. Looks like that was the wrong pick in this case.
I'll look to see if I can make a better information choice during the migration, but this is likely to be a rare situation and will only happen this one time.

aanon4, 2025-01-21:
https://github.com/aredn/aredn/pull/1834

K5RA, 2025-01-21:
I upgraded HAP-FB from NB 20250118 to NB 20250120.  The restored Dial by IP service advertisement was not changed.  However, the Installed Packages count went from six to fourteen.  The first upgrade to NB reduced it from 14 to 6, but I did not make a copy of the installed packages.  Here is that window after NB 20250120 upgrade.  

[image]

It looks like #1833 is a one-time event.  I will set up a node and try the upgrade again with next NB.  Should be able to close #1833 soon.

I am grateful for your support of AREDN.

aanon4, 2025-01-21:
Yes it's a one time upgrade. This PR will not make any difference to you because you already upgraded.

K5RA, 2025-01-27:
Let's get this Issue off the table.  I am not seeing it on my HAP-FB node as I migrate to new NBs.

closed 2025-01-27 by K5RA