aredn/aredn #1835: TP-Link 710 v1.6 on nightly 1/21 firmware fails the Ping function when the 710 is the target, works if 710 is source

alchemy33, 2025-01-21:
### Describe the bug
Logged in and using the node tools utilities between 2 nodes with RF link, the ping function fails when the CPE710 is the Target.  The Ping function performs as expected using the "swap" selection if the CPE710 is the Source.

### Expected behavior
The ping function should perform in either swap selection.  This was confirmed between 2 Mikrotik SXT 2nD nodes in exact configuration.

### Support data
I copied the resulting text from the Ping tool window into a simple text file 

### Screenshots
Text from screen pasted into file.  Config was Linux Firefox browser to local node via lan port on injector with CPE710 on the PoE port of the injector.  Single MikroTik SXT 2nD node powered 20 ft away.  Both on channel 180, 20MHz bandwidth and 7dBm power and max distance @ 2 miles and LQM off.

### Additional context
I had orginally thought this behavior was related to the Mikrotik node but I was able to duplicate the ping test between 2 Mikrotik SXT 2nD nodes with no problem observed.  I thought this because the numeric IP of the Mikrotik node was visibly decremented -1 in the Ping header description which I thought to be an error but ping works fine between Mikrotik nodes.  I included copied examples of how the ping target 

swapping function with both MikroTik nodes works fine.  

I am not a programmer or network engineer so it may be that this a problem related to the implementation of the Ping function in the node tools. I am only reporting observed behavior.  

I was able to use the ping function from a Mikrotik SXT 2nd node on the 1/21 nightly to successfully ping a Ubiquiti M2 Rocket (running 10.24.10 firmware) both as Target and Source.

[TPLinkCPE710MikrotikSXT5acJan21Ping.txt](https://github.com/user-attachments/files/18497441/TPLinkCPE710MikrotikSXT5acJan21Ping.txt)

alchemy33, 2025-01-22:
The bug report was closed by mistake.  It should be reopened but I don't have authority to correct my mistake.

aanon4, 2025-01-22:
CPE710 have problems.
You didnt really say how well this connection was performing otherwise, just that the ping failed in one direction. How is it otherwise? What does traceroute and iperf thinkj? Does the work as expected? Do other things?

closed 2025-01-22 by alchemy33