aredn/aredn #1836: "Logout" and "Reboot" disappear from the Login-Button

Orv, 2025-01-22:
Snagged from the AREDN forums.
I checked - he's right.

Hi all,
have noticed some strange behaviour after trying 20250121-b46f49a9 and the one from 20250119 on one of my hAP lites.

Have set up my oldest hAP lite to run as a WLAN-client from my phone, so that I can run demos away from home with a tunnel to my home node, so no WAN port, and no access point, just mesh and LAN.

On the nightlies from Jan 19 and 21 I can log in to the node and use all the options panels (firmware, network, local services, radio, LAN DHCP etc) as usual, however if I leave one of these panels with Cancel rather than Done, then "Logout" and "Reboot" disappear from the Login-Button, leaving only "Help" and "Website" as options.

"Logout" and "Reboot" only reappear if I leave the node-GUI and return to it via another node. Strange...
Will post support data if anybody wants it. Cheers, HB9XBO

aanon4, 2025-01-22:
https://github.com/aredn/aredn/pull/1837

aanon4, 2025-01-27:
Fixed for you?

aanon4, 2025-01-29:
Reopen if not.

closed 2025-01-29 by aanon4