aredn/aredn #1843: Upgrade HAP2 to Nightly 0122 Issues

k1ky, 2025-01-23:
### Describe the bug
Upgrade Mikrotik HAP2 from Production 3.24.10.0 to Nightly 0122 creates several issues on Mikrotik HAP2.
Node addresses are 192.168.1.x ranges instead of the 10.xx ranges as indicated in the settings.
Node will NOT revert to Production even when using Dangerous Upgrade.
Some info did not transfer, Description info however the Tunnel information did transfer.
WAN port DHCP does not acquire internet,

- [ ] Does this issue still occur on the latest version? (**Required**)
Nightly 1/22

### Expected behavior
Expected clean upgrade.

### Support data

[supportdata-KN4CCQ-HAP2-150-MBT-2025-1-23-2-28.tar.gz](https://github.com/user-attachments/files/18514330/supportdata-KN4CCQ-HAP2-150-MBT-2025-1-23-2-28.tar.gz)

Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-01-23:
Could I have you repeat the process for me. Reinstall 3.24.10.0 and then try the upgrade. I need to know if this is repeatable or not. Thanks.

k1ky, 2025-01-23:
Understood - will see what we can do.  I haven't tried any other upgrades, are you aware of any problems with other models?  I'm a little gun shy to try at this point!

aanon4, 2025-01-23:
You should always be guns shy when we're doing nightlies.
And no, but then people dont upgrade from production to a specific nightly so it easy for something to creep in over time.
I would suggest reinstalling this same hap-ac2 rather than risking another node which may suffer the same fate if this is reproducible.

k1ky, 2025-01-23:
FYI - I was able to revert to 3.24.10.0 by NOT saving the data, so fresh population of the node information in progress.  10. addys are back baby!

k1ky, 2025-01-23:
Successful upgrade to Nightly 122 from 3.24.10 production with clean data.  This unit was not behaving properly with LAN Wi-Fi before upgrade, which is the reason we tried upgrade.  It's possible that there was an underlying corruption in the existing data in the node before the initial upgrade. We ccan probably close this issue for now pending other successful upgrade results on this model.

k1ky, 2025-01-24:
No further issues - closing this issue.

closed 2025-01-24 by k1ky