aredn/aredn #1845: x86/64 VM Image Fatally Crashes During Setup

cameronzucker, 2025-01-23:
### Describe the bug
When setting up the VM on a Proxmox server that's a LAN device of another node (required for communication on Class C network LAN on Proxmox default gateway), changing any setting in the New UI after first successful boot and DtD mesh with another device will fatally cripple the VM. DtD connectivity will stop, it won't be reachable at an established LAN IP, and is only recoverable by restoring from a previous working image. The only way around this is to enter the Old UI, then uncheck the DHCP checkbox in network settings. Disabling LAN in the new UI does not avoid the issue, and 3.24.10.0 does not have the DHCP toggle in the new UI. It does not matter if more than one NIC is available to the VM. It does not matter what type of VM architecture is used (i1440fx/q35).

- [X ] Does this issue still occur on the latest version? (**Required**)
- This issue is confirmed in 3.24.10.0

### Expected behavior
The VM, after successful configuration on first boot, should establish a DtD connection with other nodes on the network. Settings should be editable without issue.

### Support data
Attaching support data from a current nightly build image and 3.24.10.0 image.

[supportdata-N7CPZ-Nightly-20250122-dc1f8d1e-2025-1-23-2-58.tar.gz](https://github.com/user-attachments/files/18518576/supportdata-N7CPZ-Nightly-20250122-dc1f8d1e-2025-1-23-2-58.tar.gz)
[supportdata-N7CPZ-3-24-10-0-Stable-2025-1-23-9-58.tar.gz](https://github.com/user-attachments/files/18518575/supportdata-N7CPZ-3-24-10-0-Stable-2025-1-23-9-58.tar.gz)

### Screenshots
No applicable screenshots available.

### Additional context
This problem was detected on a Proxmox server running PVE 8.3.3, but it doesn't seem hypervisor-specific or related. All VM configuration is done exactly as described in the wiki: https://wiki.bayareamesh.us/index.php/AREDN_on_Proxmox

aanon4, 2025-01-23:
I'm a little unclear on what's going on here, but can you repeat what you're doing but using nightly only so I have something simpler to work with - thanks.

cameronzucker, 2025-01-24:
I can screen record this process on a nightly build when I get a chance this weekend.

aanon4, 2025-01-24:
Please just rerun whatever it is you doing, but only use nightly - thanks. I dont want to re-debug stuff that might have been fixed.

cameronzucker, 2025-01-25:
This is resolved on 20250124-7b4221e9.

closed 2025-01-25 by cameronzucker