aredn/aredn #1846: qemu-ga Package Doesn't Work on Latest VM Nightly Builds

cameronzucker, 2025-01-23:
### Describe the bug
The qemu-ga package, which worked when it was first added to a nightly build and made available in the package list for the x86/64 VM, no longer functions on the latest nightly versions with the new OpenWRT version. Running a VM management command like Shutdown in Proxmox will result in the command failing with the following error:

[image]

Researching this error suggests an OS incompatibility issue.

- [X ] Does this issue still occur on the latest version? (**Required**)
Package is only available on nightly builds. Observed in nightly 20250122-dc1f8d1e.

### Expected behavior
Management commands communicating with qemu-ga should run successfully, without errors.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-N7CPZ-Nightly-20250122-dc1f8d1e-2025-1-23-2-58.tar.gz](https://github.com/user-attachments/files/18518691/supportdata-N7CPZ-Nightly-20250122-dc1f8d1e-2025-1-23-2-58.tar.gz)

### Screenshots
Please see above.

### Additional context
I believe this is simply an incompatibility issue with the new OpenWRT version. If an updated version of the package can be added, it should resolve the issue.

aanon4, 2025-01-23:
I've seen the device used to communicate with proxmox fail to be created after the package is installed, which happens after an upgrade as the packages are reinstalled. Please try rebooting and seeing if that fixes it.

cameronzucker, 2025-01-24:
Performed the following troubleshooting steps:

* Uninstalled qemu-ga
* Rebooted node
* Reinstalled qemu-ga
* Rebooted node again
* Waited a few minutes for connections to establish after boot and tried guest agents commands from Proxmox with the same results:

[image]

aanon4, 2025-01-24:
I guess the OpenWRT package is just broken at the moment. Well see if they give it any love before their release but I wouldnt expect so.

cameronzucker, 2025-01-28:
Did the following troubleshooting from a reddit thread on the subject:

   * From console "poweroff" in the VM (not from the Proxmox WebUI)

   * Goto the options and confirmed "QEMU Guest Agent" is enabled... default is disabled

   * From Proxmox WebUI, "Start" the VM

   * Check WebUI and "qm agent <vm_id> ping"

   * root@tanaka:~# qm agent 1001 ping root@tanaka:~# qm agent 1001 get-time 1725728090851264000

After that, changed qemu guest agent type to ISA in PVE 8.3.3. Communication was restored and the VM accepts management commands. This suggests it's bypassing a VirtIO driver incompatibility with the new OpenWRT version.

closed 2025-01-28 by cameronzucker