aredn/aredn #1848: x86/64 VM No Longer Displays Mesh Information On Latest Nightly Build

cameronzucker, 2025-01-23:
### Describe the bug
After upgrading a VM to the latest build (20250122-dc1f8d1e) from stable, several meshing functions no longer work:

The cloud mesh function no longer appears even though the node is directly connected to a working Supernode tunnel. Other nodes on the DtD mesh can see the Supernode and display the cloud mesh icon, can navigate through it, etc.

[image]

The local mesh display doesn't show any other nodes, DtD, RF, or otherwise:

[image]

Compare to a DtD node on the same LAN:

[image]

- [X ] Does this issue still occur on the latest version? (**Required**)
- Observed on Nightly 20250122-dc1f8d1e

### Expected behavior
Cloud mesh icon should appear when connected to a Supernode tunnel, as it does on other DtD connected nodes. Local mesh display should accurately reflect existing connections.

### Support data

[supportdata-N7CPZ-Nightly-20250122-dc1f8d1e-2025-1-23-2-58.tar.gz](https://github.com/user-attachments/files/18519033/supportdata-N7CPZ-Nightly-20250122-dc1f8d1e-2025-1-23-2-58.tar.gz)

### Screenshots
See above.

### Additional context
This is not thought to be a network issue due to the two test VMs sharing the same switch ports with the same VLAN configuration. The node running the latest Nightly build is acting as the DHCP server and router for all other DtD nodes, and is running the tunnel server. Other nodes can connect to it and its LAN services as normal.

aanon4, 2025-01-23:
The support data shows all the nodes in the underlying files so not sure why the display isnt capturing them.
Let's upgrade to the current nightly and repeat the test.

aanon4, 2025-01-23:
Looks like there's a new race condition between the network starting up and olsr starting up.

aanon4, 2025-01-23:
Unclear why this is happened, so for the moment I'm including a connectivity check in the olsrd watchdog.
https://github.com/aredn/aredn/pull/1852

cameronzucker, 2025-01-24:
Resolved in 20250123-a55b053e.

[image]

[image]

closed 2025-01-24 by cameronzucker