aredn/aredn #1849: Make x86/64 VM Setup More Friendly

cameronzucker, 2025-01-23:
**Is your feature request related to a problem? Please describe.**
During setup, the only user configuration taking place inside the VM itself consists of the following after first boot (from the Wiki because I'm lazy):

```
From the console execute the following command:

/usr/local/bin/firstuse-setup <NODE NAME> <PASSWORD>

Where <NODE NAME> should be replaced by your node's new name, and <PASSWORD> should be a password to protect access to this node. For example:

/usr/local/bin/firstuse-setup KN6PLV-MyNewNode mysecretpassword
```
This works fine for setting the node attributes, but there's no output of the node's IP address after it's been named like there is in the normal setup. This makes troubleshooting more difficult since obtaining the IP relies on either a working DtD link or interpreting the output of "ip addr show", which is far from concise.

**Describe the solution you'd like**
Please modify the firstuse-setup script to output the assigned LAN IP after the node's been named, similarly to how the web UI displays the assigned IP for traditional nodes during first time setup.

**Describe alternatives you've considered**
This is the easiest implementation I could think of. Obtaining the IP can be very frustrating for new VM users who are not particularly network savvy, especially if they have even a minor VM, hypervisor, or network config issue preventing local communication and are relying on the hypervisor console.

**Additional context**
None.

cameronzucker, 2025-01-25:
firstuse-setup <args> also seems to oscillate between "firstuse-setup" and "firstuse_setup" on different builds.

aanon4, 2025-01-28:
https://github.com/aredn/aredn/pull/1861

aanon4, 2025-01-31:
Please verify this change - thanks

aanon4, 2025-02-04:
Please verify this is fixed.

cameronzucker, 2025-02-04:
Feature confirmed working as requested. Thanks for the very quick implementation.

[image]

closed 2025-02-04 by cameronzucker