aredn/aredn #1850: .local.mesh TLD Registration Potential Issue

cameronzucker, 2025-01-23:
### What is your question
This is not a problem today, but could easily be in the future.
[Krebs article on this issue](
https://krebsonsecurity.com/2024/08/local-networks-go-global-when-domain-names-collide/)

Right now, .local.mesh or just .mesh are not publicly available Top Level Domain (TLD), but with the addition of .zip, .mp4, .lol, etc., it easily could be. This would create a similar situation to the one described in the article. While AREDN networks are primarily isolated from WAN, users navigating to .local.mesh addresses on internet-capable machines may make errant DNS requests to outside servers. It's a potential conflict at best and a hijacking issue at worst.

At this stage of the project, moving away from .local.mesh is probably infeasible due to the devices this would orphan on older firmware immediately after the change is implemented. AREDN should be keeping an eye on the IANA registration windows to potentially scoop up .local.mesh if it ever becomes publicly available.

aanon4, 2025-01-23:
A gTLD, which .mesh would be, costs a quarter of a million dollars just to register with running costs on top of that. Even if we qualified for the charity discount, the costs is still tens of thousands. Not really a good use of money.

closed 2025-01-23 by aanon4

cameronzucker, 2025-01-24:
I believe this was misunderstood.

Rather than suggesting the AREDN project preemptively make available and purchase the .mesh gTLD, there should just be an awareness of this very real potential issue should .mesh become available. If that happens, local.mesh could be scooped up for a very modest cost since it would the be the same as any other niche domain registration.