aredn/aredn #1851: List of nodes is empty after nightly build upgrade.

Orv, 2025-01-23:
### Describe the bug
After upgrading to nightly build 20250123-a55b053e, the "Search the mesh" screen is blank except for the local node.

Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
Should show a list of network nodes, sorted by quality.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-W6BI-SHACK-NODE-2025-1-23-20-6.tar.gz](https://github.com/user-attachments/files/18526411/supportdata-W6BI-SHACK-NODE-2025-1-23-20-6.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

### Additional context
Note that all links are still functional; it's just the display of them that is missing.

aanon4, 2025-01-23:
Duplicate of https://github.com/aredn/aredn/issues/1848 I think

Orv, 2025-01-23:
Looks like it, yeah.    I'll close this one

closed 2025-01-23 by Orv

aanon4, 2025-01-23:
Unclear why this is happened, so for the moment I'm including a connectivity check in the olsrd watchdog.
https://github.com/aredn/aredn/pull/1852