aredn/aredn #1853: PowerBeam PBM5 doesn't resolve nodes during Traceroute and Ping Tests

k1ky, 2025-01-24:
### Describe the bug
Traceroute and Ping tests don't resolve node connections under test.  Just shows recurring string of * instead of node names.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes, Nightly 123 and previous.
### Expected behavior
A clear and concise description of what you expected to happen.
Expect test to complete instead of running a string of * * *
### Support data

[supportdata-K7DCC-PBEM5XW-3-VTSS-245-2025-1-23-19-17.tar.gz](https://github.com/user-attachments/files/18528972/supportdata-K7DCC-PBEM5XW-3-VTSS-245-2025-1-23-19-17.tar.gz)

Please add the support data for all nodes mentioned in this bug report (**Required**)
Attached above.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
This is a test to a neghboring node 5 miles away with a good path.
[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Iperfspeed to this neighbor works, I am able to access the node GUI, perform Firmware updates, etc.  I have other PBM5XW units that work fine. Problem is isolated to this node only.   It is available online our system to test.

aanon4, 2025-01-24:
What about other node pairs? Is it everything or just this specific path? Also, please add the nodes in the text so I can try them myself (rather than trying to get them from an image). Thanks.

k1ky, 2025-01-24:
The is the node that it talks to: http://w4ghd-rkm5xw-109-lavergne-100twr-345.local.mesh/a/status
Everything works in reverse from the target node. It wont resolve any other paths either from the k7dcc node
Do you need the admin password?

k1ky, 2025-01-24:
Same result with Nightly 124

aanon4, 2025-01-24:
This should fix it:
https://github.com/aredn/aredn/pull/1856
However, can you do the following as a test with your current setup? Can you plug something into the ethernet of the PowerBeam as well as just power. Looks like there's no active network there which is probably the root of the problem. This fix will take care of this, but it would be good to verify the root cause.

k1ky, 2025-01-24:
Unable on this unit - I have others with no Ethernet connection that work fine.  Unit is at a facility with limited access.  Would be next week before I could test, but will do.

aanon4, 2025-01-24:
That's okay. If it's not handy then please dont bother.

k1ky, 2025-01-25:
Nightly 125 did the trick!  Closing issue.

closed 2025-01-25 by k1ky