aredn/aredn #1855: Service installation

csmaslin, 2025-01-24:
I haven't been able to implement a service to my NAS that hosts my webiste successfully over the last 5 updates.

Attempts to reach the device/service , the system reports tha the path/url cannot be found.

http://10.137.85.101.local.mesh/index.htm.

address reservation=10.137.85.101

Add service
Add a service from a template

Generic URL
+
website

-http://10.137.85.101:80/index.htm
-
Host aliases
DNS hostname aliases
+
website 10.137.85.101
-
Port Forwarding
Forward WAN port to LAN

aanon4, 2025-01-24:
Please add the support data for this node.
When you say "over the last 5 updates" this suggests this use to work - is that the case?

csmaslin, 2025-01-24:
The nightly builds and yes it did use to work.

Chuck

Orv, 2025-01-24:
Chuck, I'll work with you on this.
Orv W6BI

closed 2025-01-25 by csmaslin