aredn/aredn #1857: Neighborhood node list indicates 50% errors erroneously

AI7NC, 2025-01-26:
### Describe the bug
Six of our tunnel clients on a core tunnel server node show an error rate of 50% while the others indicate 0%. We have no inication of packet loss with either ping, MTR, or iperf3 tests. This node is in a peering facility with direct gigabit ethernet connectivity to an ISP's network. 

- [ ] Does this issue still occur on the latest version? Yes, running 3.24.10.0

### Expected behavior
I would expect all tunnels to be 0%, if there was error rates I'd expect to see a varied rate of error between the tunnels and not a black/white 0/50. 

### Support data
Support attached

### Screenshots

[image]

[image]
[image]
[image]
[image]

aanon4, 2025-01-26:
What version are you running?

aanon4, 2025-01-26:
I should add that there was a bug (fixed fairly recently) where LQM could get confused about which tunnel was on which network device. The symptoms of this are like you describe here.

AI7NC, 2025-01-26:
> What version are you running?

[image]

aanon4, 2025-01-26:
With reference to the comments above, please retest on nightly.

AI7NC, 2025-01-26:
> I should add that there was a bug (fixed fairly recently) where LQM could get confused about which tunnel was on which network device. The symptoms of this are like you describe here.

LQM is disabled. Would the bug still be exhibited?

[image]

AI7NC, 2025-01-26:
> With reference to the comments above, please retest on nightly.

We won't run a nightly on this node, it's too important to our network. Can you point me to the pull request you think resolves this?

We can keep an eye on our less important nodes if they exhibit this problem, unfortunately it usually shows up weeks after a reboot.

aanon4, 2025-02-04:
Please verify this is fixed.

AI7NC, 2025-02-05:
I cannot verify the fix until a stable release is made. We don't run
nightlies on this node.

To add complication to verification, the issue only shows up after weeks of
on time. I rebooted the node 9 days ago and the issue hasn't shown back up
yet. It will eventually but I don't understand the mechanism that triggers
it. I have seen this issue multiple times though over the past year with
more than one stable firmware release.

73,

Nathan Campbell, AI7NC

closed 2025-02-05 by aanon4