aredn/aredn #1860: x86/64 VM Can't See Supported Wi-Fi Radios

cameronzucker, 2025-01-28:
### Describe the bug
OpenWRT should support Mediatek and other Wi-Fi cards when the PCIe device and matching USB interface are passed through to the VM from the hypervisor.

The following settings were changed in PVE 8.3.3 to enable the Mediatek 7922 Wi-Fi 6E card:
* VM type changed to Q35 for PCIe
* iOMMU enabled (Intel - VirtIO fatally crashes the VM)
* ROM-Bar was toggled on/off to see if it made a difference
* PCIe device was passed both with and without full functionality
* USB device and port were both tried

[image]

[image]

[image]

[image]

However, after all of this and installing every available supporting package with any relevance, the VM still doesn't recognize any installed radios:

[image]

- [ X] Does this issue still occur on the latest version? (**Required**)
Observed in nightly 20250125-c6379b24.

### Expected behavior
Once the PCIe and USB devices are passed through to the VM they should appear as configurable radios.

### Support data

[supportdata-N7CPZ-M2PoEVM-2025-1-28-11-5.tar.gz](https://github.com/user-attachments/files/18577574/supportdata-N7CPZ-M2PoEVM-2025-1-28-11-5.tar.gz)

### Screenshots
Inline in bug description.

### Additional context
I haven't tried a UEFI BIOS, but as far as I know, the current image only supports legacy BIOS (SeaBIOS).

aanon4, 2025-01-28:
No plans to support radio in virtual environments.

cameronzucker, 2025-01-28:
Can we make a case for this? In addition to adding Wi-Fi as LAN, it would allow for the use of devices like this to open up new possibilities on 900 Mhz in difficult terrain: https://store.rokland.com/products/alfa-network-ahm27292u-ieee-802-11ah-sub-1-ghz-standalone-m-2-ngff-b-key-3042-type-linux-module?_pos=20&_sid=28dc8d166&_ss=r

Several regions in Arizona along the Mogollon Rim haven't adopted AREDN because of foliage preventing clean line of sight. Using the new HaLow standard with high gain antennas would allow usable speed data links in completely new areas. We'd also like to experiment with this to try establishing long distance links which are just barely obstructed on higher bands.

cameronzucker, 2025-01-28:
I was thinking the two specified WLAN cards are not too different in terms of implementation [since they have existing driver packages.](https://openwrt.org/docs/techref/driver.wlan/mt76) That may be due to my inexperience with the project and how hardware is incorporated.

What specific work needs doing? I'm more than willing to help implement this if it's within my ability.

aanon4, 2025-01-28:
The starting point is being able to build and run your own images from the source. The instructions are out of date I'm afraid so you'll have to wade through dependencies to get it working. I run builds on Debian 11. If document what you need to do to get builds up I'll update the docs so others dont have to repeat this process.

cameronzucker, 2025-02-03:
Working on this. It can be marked needs hardware. I'm limiting support to just the Mediatek 7915/7916 chips, but they'll take a bit to come in from Taiwan. The HaLow stuff will depend on how difficult back porting the drivers are.