aredn/aredn #1864: Add option to allow specific nodes to reach the Internet over the AREDN host

KD4WLE, 2025-01-29:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
A simple method in the GIU to add specific nodes to reach the Internet over the AREDN host with the ablity to use a wildcard (IE: KD4WLE* would allow any node starting with KD4WLE)

**Describe alternatives you've considered**
Specific instruction to add the firewall entries manualy

convolved-KY4G, 2025-02-02:
It might be pretty easy to guess a node name that would allow someone else to get the access you may be trying to exclude.  Perhaps MAC addresses would be a better alternative?  That wouldn't work well with the wildcards though.  MACs can be cloned, but it seems like it would be harder to determine than node names.

aanon4, 2025-02-03:
At a low level you are essentially asking the routing on each node to forward traffic not just based on destination but also on source. This is supported in IPv6 which we dont support, and "kind of" supported IPv4. It isn't supported by OLSR which is how we manage routes.

You could build a tunnel over the mesh (not to be confused with our current tunnels) between a gateway and the authorized node to achieve this.

But, aside from the complexity in doing this in a way to hide that complexity, you didnt provide a use case for this and I'd be interested in that.

KD4WLE, 2025-02-03:
For example, We have a number of remote nodes at a DMR repeater site, We use AREDN to link these with to a node with internet connectivity and brandmeister.

aanon4, 2025-02-03:
But you already do that. I'm curious what this specific mechanism brings you.

KD4WLE, 2025-02-03:
This was into Orv’s recent email.. IMHO, if it isn't a pending issue with Dev, it isnt an issue with me.

jlp78, 2025-02-15:
Here's a use case where it would be desirable... 

Let's say I have several nodes that I control.  One of them has a WAN connection to the internet.  This node can easily install firmware updates or packages over the WAN.  Other nodes that I control can't and I must manually download new firmware and push it to them (or run an on-mesh repository of releases and packages, which arguably is the right answer for this).

I would be satisfied with a way to allow DTD connected nodes to access the internet via my WAN connection, similarly to how I can allow my LAN connected hosts.