aredn/aredn #1867: Mikrptik hAP3 GPS data not seen

wu2s, 2025-01-29:
### Describe the bug
A clear and concise description of what the bug is.
Earlier night;y builds (december 2024) saw the GPS device and set the time from GPS data. Over the last 2 weeks or so of nightly builds, this function stopped working.
- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior USB port and set the time via this device. Th
A clear and concise description of what you expected to happen.
With the whereandwhen library loaded. the Mikrotik should see the GPS data instead of using NTP.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-WU2S-H3-X-2025-1-29-12-26.tar.gz](https://github.com/user-attachments/files/18592388/supportdata-WU2S-H3-X-2025-1-29-12-26.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-01-29:
The logs say that you upgraded the node, the node reinstalled whereandwhen, but it's not running. If the GPS device isn't plugged in please make sure it is, and then reboot your node. Let it run for 5 minutes, then capture the support data and post it here.  Thanks.

Also - what version GPS dongle are you using?

wu2s, 2025-01-29:
GPS unit is the BU353N. The GPS device is powered on if the red LED on it is a power on indicator.
I will reboot the node and wait.

wu2s, 2025-01-29:
Po

[supportdata-WU2S-H3-X-2025-1-29-17-16.tar.gz](https://github.com/user-attachments/files/18595342/supportdata-WU2S-H3-X-2025-1-29-17-16.tar.gz)

st reboot support data attached

aanon4, 2025-01-29:
Well the kernel doesn't find the gps device which appears as a serial line in /dev. Possibly we need additional kernel driver for this device with the newer kernels we now run.  I'll see if I can work out if that's the case and what they might be.

aanon4, 2025-01-30:
Looks like the installed driver should be working.

Do you have a powered USB hub? If you have, we can power the USB device that way and see if this is a power problem or something else.

wu2s, 2025-01-30:
I think I have a powered USB hub here somewhere. Searching ...

wu2s, 2025-01-30:
Updated to NB 2025-01-50 and waited until packages were reinstalled. Now using a powered USB hub too.

[supportdata-WU2S-H3-X-2025-1-30-9-4.tar.gz](https://github.com/user-attachments/files/18604602/supportdata-WU2S-H3-X-2025-1-30-9-4.tar.gz)

aanon4, 2025-02-02:
I'm adding the loadable package `usbutils` to the build (see https://github.com/aredn/aredn/pull/1875). Could you try this build tomorrow, install the package, and then use the command `lsusb` so we can see what usb devices the node can see. The logs still dont show anything plugged in.

wu2s, 2025-02-02:
Yes, will do.

wu2s, 2025-02-02:
Installed lattest NB 20250202, the  installed usbutils package.
SSH to the device.
Command lsusb gave an error:
unable to initialize libusb: -99

[supportdata-WU2S-H3-X-2025-2-2-11-23.tar.gz](https://github.com/user-attachments/files/18633235/supportdata-WU2S-H3-X-2025-2-2-11-23.tar.gz)

aanon4, 2025-02-02:
Could you try the following?
```
export LIBUSB_DEBUG=9
lsusb
```
It might give us more to go on.

wu2s, 2025-02-02:
root@WU2S-H3-X:~# export LIBUSB_DEBUG=9
root@WU2S-H3-X:~# lsusb
unable to initialize libusb: -99
root@WU2S-H3-X:~#

aanon4, 2025-02-03:
More things to try (although I'm expecting this to also produce no output):
```
cat /sys/kernel/debug/usb/devices
```

aanon4, 2025-02-03:
I'm guessing it needs a different USB controller, so I've added more to the firmware build (tomorrow's nightly) as well as adding more to the WhereAndWhen package. If you want to try loading up the nightly then the new WhereAndWhen and see where that gets us, that'd be great.

wu2s, 2025-02-03:
cat /sys/kernel/debug/usb/devices produced no output.

wu2s, 2025-02-03:
root@WU2S-H3-X:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.73 xhci-hcd xHCI Host Controller
Bus 001 Device 002: ID 0bda:5411 Generic USB2.1 Hub
Bus 001 Device 003: ID 0bda:5411 Generic USB2.1 Hub
Bus 001 Device 004: ID 067b:23a3 Prolific Technology Inc.  USB-Serial Controller 
Bus 001 Device 005: ID 0bda:5411 Generic USB2.1 Hub
root@WU2S-H3-X:~# 

root@WU2S-H3-X:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=01 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.06
S:  Manufacturer=Linux 6.6.73 xhci-hcd
S:  Product=xHCI Host Controller
S:  SerialNumber=xhci-hcd.0.auto
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 4
D:  Ver= 2.10 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=5411 Rev= 1.01
S:  Manufacturer=Generic
S:  Product=USB2.1 Hub
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=01 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms
I:* If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms

T:  Bus=01 Lev=02 Prnt=02 Port=00 Cnt=01 Dev#=  3 Spd=480  MxCh= 4
D:  Ver= 2.10 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=5411 Rev= 1.01
S:  Manufacturer=Generic
S:  Product=USB2.1 Hub
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=01 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms
I:* If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms

T:  Bus=01 Lev=03 Prnt=03 Port=00 Cnt=01 Dev#=  5 Spd=480  MxCh= 4
D:  Ver= 2.10 Cls=09(hub  ) Sub=00 Prot=02 MxPS=64 #Cfgs=  1
P:  Vendor=0bda ProdID=5411 Rev= 1.01
S:  Manufacturer=Generic
S:  Product=USB2.1 Hub
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=01 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms
I:* If#= 0 Alt= 1 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=02 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   1 Ivl=256ms

T:  Bus=01 Lev=02 Prnt=02 Port=03 Cnt=02 Dev#=  4 Spd=12   MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=067b ProdID=23a3 Rev= 1.05
S:  Manufacturer=Prolific Technology Inc. 
S:  Product=USB-Serial Controller 
S:  SerialNumber=CRCIe108Y20
C:* #Ifs= 1 Cfg#= 1 Atr=a0 MxPwr=100mA
I:* If#= 0 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=pl2303
E:  Ad=81(I) Atr=03(Int.) MxPS=  10 Ivl=1ms
E:  Ad=02(O) Atr=02(Bulk) MxPS=  64 Ivl=0ms
E:  Ad=83(I) Atr=02(Bulk) MxPS=  64 Ivl=0ms
root@WU2S-H3-X:~# 

[supportdata-WU2S-H3-X-2025-2-3-8-42.tar.gz](https://github.com/user-attachments/files/18643820/supportdata-WU2S-H3-X-2025-2-3-8-42.tar.gz)

aanon4, 2025-02-03:
Well this looks a lot better. The GPS USB serial device is being created. I dont see the gps daemon running, but possibly you got the support data before that happened. It can take 10 minutes or so depending on when things start up. Can you take another look?

wu2s, 2025-02-03:
Here is the latest support data

[supportdata-WU2S-H3-X-2025-2-3-11-49.tar.gz](https://github.com/user-attachments/files/18646206/supportdata-WU2S-H3-X-2025-2-3-11-49.tar.gz)

aanon4, 2025-02-03:
Looks like GPSD has started up correctly. Next questions ... is it actually working?

wu2s, 2025-02-04:
How do I determine if the GPSD is actually working?

aanon4, 2025-02-04:
Unless you disabled this (it's enabled by default) there should be a little (gps) label next to the time on the main status page.

wu2s, 2025-02-04:
showing ntp.

[image]

Nope. Still

aanon4, 2025-02-04:
Is the gps received somewhere it can see gps satellites?

wu2s, 2025-02-04:
Yes it is in a Southwest facing window.

aanon4, 2025-02-04:
Can you install the package 'gpsd-clients'. This contains some tools we can use to check the gps is doing what it should. I think the one you need after that is 'gpspipe' which will dump what the gps is doing.

wu2s, 2025-02-04:
I downloaded the package from https://downloads.openwrt.org/releases/packages-23.05/x86_64/packages/gpsd-clients_3.25-1_x86_64.ipk
but it does not load.
Does it need to be compiled specifically for the Mikrotik device?

wu2s, 2025-02-04:
Oddly the AP ac3 does not get an updated package list.
But a hAP ac2 does obtain a package list.

[image]

aanon4, 2025-02-04:
Could you give me login on your hap-ac3?

wu2s, 2025-02-04:
Yes. Sent via email

aanon4, 2025-02-04:
Not sure which email, but tim.j.wilkinson@gmail.com works.

aanon4, 2025-02-05:
So I was able to watch what the sensor produces, and it says it cannot find any satellites. Maybe unpowered it and reboot the node (not sure jut rebooting the node will unpowered the sensor) is worth trying? Any chance you can give it a view of the sky outside (I'm sure it's not warm or dry right now)? Given this the system is behaving correctly ... but not in a useful way.

wu2s, 2025-02-12:
Tested the GPS receiver dongle on a windows computer with GPSView. The donge works correctly.
Installed release 3.25.2.0 on the hAPac3 and confirmed that all needed packages were installed.
Plugged in the GPS USB receiver and waited for a few minutes.

The devices are now working correctly.
Source of the bug never determined but let's consider this issue closed.

closed 2025-02-12 by wu2s