aredn/aredn #1868: Hosted services DNS aliases aren't propgating outside of node LAN

cameronzucker, 2025-01-31:
### Describe the bug
It's unclear whether this problem is isolated to the new OpenWRT version nightly builds, but hosted services to not propagate DNS properly through the mesh. They are accessible through the mesh via IP address, or on the local node LAN via DNS alias, but not via the mesh using DNS alias. Being reachable by IP:port suggests this isn't a network or port forwarding issue and is isolated to DNS resolution.

Services are configured as follows with globally unique hostnames:

[image]

Tested the following conditions:
* Node directly connected to a second node via DtD/LAN. Second node acting as primary router and DNS server.
* Node connected to larger network via Wireguard tunnel. Network entry node also connected to the same cloud mesh over Wireguard tunnel.

Results were the same - services fail to resolve by DNS alias, but can be reached by direct IP:port.

nslookup returns no results for any of them:

[image]

[image]

HAR trace while accessing a test service only shows DNS resolution failure:

[image]

Shelled into N7CPZ-M2PoEVM and ran `/etc/init.d/dnsmasq restart`, then left the node powered on and connected to the mesh for 6+ hours to see if DNS propagated, but there was no change.

- [X] Does this issue still occur on the latest version? (**Required**)
Issue was observed in nightly 20250130-8dd59784

### Expected behavior
After a reasonable amount of time, DNS should propagate through the mesh and services should be reachable by DNS alias if using a globally unique hostname.

### Support data
N7CPZ-M2PoE is the node hosting services unreachable by DNS alias. N7CPZ-HalfRackVM is the node the services are not reachable from by alias during testing.

[supportdata-N7CPZ-M2PoEVM-2025-1-30-18-22.tar.gz](https://github.com/user-attachments/files/18611637/supportdata-N7CPZ-M2PoEVM-2025-1-30-18-22.tar.gz)

[supportdata-N7CPZ-M2PoEVM-2025-1-30-18-22.tar.gz](https://github.com/user-attachments/files/18611672/supportdata-N7CPZ-M2PoEVM-2025-1-30-18-22.tar.gz)

### Screenshots
Please see above.

### Additional context
None.

aanon4, 2025-01-31:
Looks like you uploaded the same support data twice, rather than the other node.

aanon4, 2025-01-31:
Also could you attached a screenshot of the DHCP allocations - thanks.

cameronzucker, 2025-01-31:
So I did - sorry about that. Corrected.

DHCP allocations:

[image]

aanon4, 2025-01-31:
Mind you, our documentation says it should work ... so i guess I need to work out which one is wrong.

cameronzucker, 2025-01-31:
The way it was written I figured "do not propagate" referred to the reserved hostname, rather than the alias registered in a different menu. I can re-test in a bit.

aanon4, 2025-01-31:
Documentation is correct ... the code is broken. Thanks. Will fix.

aanon4, 2025-01-31:
https://github.com/aredn/aredn/pull/1869

aanon4, 2025-02-04:
Please verify this is fixed.

cameronzucker, 2025-02-04:
This confirmed resolved on 20250203-5eb740a8.

closed 2025-02-04 by cameronzucker