aredn/aredn #1871: Rethinking the use of DYN_GW plugin

mathisono, 2025-02-01:
give a node as way to determined what exit gateway it's used to reach the internet, and report that node's name in the json data.

"default gateway" currently selected best path by olsrd, this is currently working!
"gateway node" never implemented, I'm suggesting the use of the original DYN_GW plugin, until olsr is depreciated from ARDEN. 

pro: 
Right now with no feed back from nodes on a network, it impossible to quickly determine how nodes on a network are using different exit gateways. node info reports if its a GW but without the DYN_GW plugin function, a node just passes traffic to its default gateway until it hits an exit node. Im fine with that, but I caution mode advanced determination out side what DYN_GW plugin provides, olsr 0.6.0 is no longer under development, OLSR v2 and BATMAN provide a better frame work to do more!!!! My intent is only to give network mappers access to display and diagnose the chosen exit GW, reported via json. https://github.com/kn6plv/NewMeshMap/issues/18

[ReadMe](https://github.com/OLSR/olsrd/blob/master/lib/dyn_gw/README_DYN_GW)

[olsrd-mod-dyn-gw-plain](https://openwrt.org/packages/pkgdata_owrt18_6/olsrd-mod-dyn-gw-plain) 

cons: 
I don't know how compatible with the current AREDN source, the old code is? [LINK](https://github.com/OLSR/olsrd/tree/master/lib/dyn_gw/src)

aanon4, 2025-02-02:
Could you describe the goal here. I understand what it does, how it does it, and how it's different from now; but I dont know what specific problem it is solving.
(This information may be in other channels but please add it here - thanks).

aanon4, 2025-02-20:
Moving to Babel.

closed 2025-02-20 by aanon4