aredn/aredn #1873: NTP Server Names and clock update frequency broken in latest nightly builds

FredK27, 2025-02-01:
### Describe the bug
I cannot configure a second NTP server name or set the time update frequency to "hourly". Continous and daily updates can be choses.

- [YES] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
When I set the second NTP server name, I expect to see a request to commit the change and the NTP URI should be saved. I do not see an commit request after entering a new value for the second NTP server.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**) - attached

[supportdata-AB1OC-Hollis3-NH-TGW-2025-2-1-11-19.tar.gz](https://github.com/user-attachments/files/18629477/supportdata-AB1OC-Hollis3-NH-TGW-2025-2-1-11-19.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

[image]

Can't change either the second NTP server name or the NTP Updates to "hourly:

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here. I am seeing this on three nodes here that are all running the latest nightly build (2025-02-01).

aanon4, 2025-02-01:
https://github.com/aredn/aredn/pull/1874

FredK27, 2025-02-01:
Wow, you are awesome! I am new to this process. Do I need to wait until later today to see a new build that I can use to verify the fix for you?

aanon4, 2025-02-01:
Easy to fix when I make dumb mistakes :-)
Yes, a new build with these changes will be available tomorrow. Thanks for reporting this.

FredK27, 2025-02-01:
Sure thing Tim – glad to have the opportunity to help. Thanks for all of your good work on AREDN!

Best and 73,

Fred Kemmerer,  AB1OC

Mail: ***@***.******@***.***>
ARISS Mentor and Ground Station

FredK27, 2025-02-01:
Quick question. What time of day do the nightly builds run?

Best and 73,

Fred Kemmerer,  AB1OC

Mail: ***@***.******@***.***>
ARISS Mentor and Ground Station

aanon4, 2025-02-01:
They're available about 12:00 noon UTC, and the build starts a couple of hours before that.

FredK27, 2025-02-01:
Thank you. I will test the fix later today when the build is done.

aanon4, 2025-02-01:
Jolly good. For clarification, 12 noon UTC is about 18.5 hours from now as I type this.

FredK27, 2025-02-01:
Got it

Best and 73,

Fred Kemmerer,  AB1OC

Mail: ***@***.******@***.***>
ARISS Mentor and Ground Station

FredK27, 2025-02-02:
Problem fixed and verified correct operation on three different MikroTIk nodes here. Thank you!

closed 2025-02-02 by FredK27