aredn/aredn #1876: Installing nightly 20252019bf147be causes Error

KenHorse, 2025-02-02:
### Describe the bug
Installing nightly 20252019bf147be causes "Error: Unknown error" to be displayed on x86/64 QEMU

- [ ] Does this issue still occur on the latest version? (**Required**)
As above

[image]

### Expected behavior
Should install without error!

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots

If applicable, add screenshots to help explain your problem.

### Additional context
This is a ProxMox VM

aanon4, 2025-02-02:
Usually this is network related. Try again later or just download directly and upload.

aanon4, 2025-02-10:
No follow up - closing.

closed 2025-02-10 by aanon4