aredn/aredn #1877: Nightly not allowing port 3001?

brontidebrontide, 2025-02-02:
I have a Raspberry Pi connected to my hAPac2.  The Pi has various services such as Apache, SSH, and MotionEye, all running nicely.  I also have "topologrjs version 0.1b5" (https://github.com/captainwasabi/topologrjs) running with URL access on port 3001.

Sometime after nightly 20250123-a55b053e, access to that URL on port 3001 stopped working.

I had been running the latest nightly, even today's (20250202-f1a31e06) but they all fail.

Unrelated, I lost my copies of nightlies after 20250123 and before 20250202 so I cannot tell exactly when this broke.

This was initially reported on the AREDN forum: https://www.arednmesh.org/content/nightly-not-allowing-port-3001#comment-24480

aanon4, 2025-02-02:
Please add support data for the node - thanks.

brontidebrontide, 2025-02-03:
[supportdata-AB4YY-services-2025-2-2-20-3.tar.gz](https://github.com/user-attachments/files/18635004/supportdata-AB4YY-services-2025-2-2-20-3.tar.gz)

I installed version 20250202-f1a31e06 which has the problem, and attached the support data.

aanon4, 2025-02-03:
I meant to ask, when you say "access to that URL on port 3001 stopped working." you didnt say where you were accessing the URL from. Another LAN device, another node on the mesh, somewhere else?  Also, can you ping the device but not access the URL?

aanon4, 2025-02-03:
I see there's a bunch of stuff attached; I take it the rest works fine?

brontidebrontide, 2025-02-03:
The service is on a Pi 4 named "GHF".  I can access other services:

- **BPQ** is running on the Pi: I can access port 8000 HTML access is all good.

- **Meshchat** on the same Pi is accessible.

- **Motioneye** runs on the same Pi: Port 8765 HTML access is all good.

- '**topologrjs**' runs on the same Pi: and is the only service that I can no longer access with the newer nightlies.  Going back to the older nightly always make it work again.

My browser is on a Windows 11 PC via a WiFi dongle to the hAPac2.  I've tried Brave and Waterfox browsers and both fail to bring up that :3001 page with the recent nightlies.

From the Windows 11 PC, I can use RealVNC to access the same Pi and bring up a browser and access all the pages except for the 'topologrjs' one.

I hope this provides the data you are asking for.

- Mike

aanon4, 2025-02-03:
I looked at the application itself, topologrjs, and it is using a service on the node's port 9090 which we recently locked down so it couldnt be accessed except on the node itself. The goal was to stop random nodes on the mesh accessing it (excessive calls to this port will case OLSR to crash), but locking out LAN access was a bit draconian. So I suspect that the reason port 3001 isn't accessible is because this app isn't running, or at least running correctly now. Could you check on its operational status?

I will look at restoring LAN access, but keep Mesh access restricted.

aanon4, 2025-02-03:
Looking into this a little more, the restriction was put in place because OLSR would no longer startup correctly without it. I'm guessing there are some kernel network changes which affect it (it's a very old program) so we'll have to investigate much further before attempting to restore non-local access. I'll update this Issue when there's any relevant change.

brontidebrontide, 2025-02-03:
This is a bit out of my knowledge area.  :) 

For what it's worth, on the Pi where topologrjs is running:

```
sudo netstat -natpe:
Proto  Recv-Q  Send-Q Local Address Foreign Address State   User  Inode  PID/Program name
tcp6        0       0 :::3001       :::*            LISTEN  1001  18355  1028/node /home/pi/

sudo netstat -tlp:
Proto Recv-Q Send-Q Local Address  Foreign Address  State   PID/Program name
tcp6       0      0 [::]:3001      [::]:*           LISTEN  1028/node /home/pi/
```

Even without doing anything to the Pi, I can go back and forth between today's AREDN nightly and an earlier one like 20250123-a55b053e and the web page will be dead or come alive with it only being alive with the 20250123-a55b053e (and earlier) versions.  I do not have to restart anything on the Pi, and have previously rebooted it to no effect on this problem.  Let me know if there is any more I can do to help.  Thanks.

brontidebrontide, 2025-02-03:
Okay, I'll give it a good but on another Pi that I use for testing.  I'll see if I can get that tested today.

brontidebrontide, 2025-02-03:
I installed your version of topologrjs and verified it was working, but I was on the old 20250123 firmware.  Then I installed today's firmware, and I was back to not accessing topologrjs with the browser.  And yet another test, I reinstalled the old 20250123 firmware and sure enough, topologrjs is accessible again.

brontidebrontide, 2025-02-03:
Wait-a-sec....  I was on your fork but it contained the same nice and thorough installation instructions which did a git : https://github.com/captainwasabi/topologrjs.git  !!

So, I'll have to delete all that and use your fork.  Ha, simple mistake....I suppose.

aanon4, 2025-02-03:
If it still fails, can you send me a screenshot of your browser so I can see the exact fail message? Also, I'd quite like to see the result of using CURL to access the port, but I'm not sure what the Windows equivalent is for that.

brontidebrontide, 2025-02-03:
Success!  Using your version of topologrjs works fine.  I suppose that is the best solution?

aanon4, 2025-02-03:
Until the original owner accepts my PR, probably yes.

closed 2025-02-03 by aanon4

aanon4, 2025-02-04:
FYI. The owner of topologrjs has merged my change, so you can now use the original repo again.