aredn/aredn #1887: New UI Overly Restricts URL Schemes using Generic URL Template

AI7BQ, 2025-02-09:
### Describe the bug
Beginning with firmware version 3.24.10.0, Local Services added with the Generic URL Template using the URL Scheme 'w2lk' are not permitted. The issue occurs on all nightly builds through 02-08-2025.

### Expected behavior
I expect the URL Scheme field to remain unvalidated, consistent with the behavior in the old UI.

### Workaround
Use the old UI.

aanon4, 2025-02-09:
I dont know what "URL Scheme 'w2lk'" is. Can you expand on this?

aanon4, 2025-02-09:
Do you mean a url such as ```w2lk://test.com/stuff```?

aanon4, 2025-02-09:
https://github.com/aredn/aredn/pull/1888

aanon4, 2025-02-09:
The above chance means the url scheme / protocol is now validated to the spec (w2lk is valid)

AI7BQ, 2025-02-09:
If #1888 covers the issue, then this issue can be closed. Yes, I do mean wl2k://test.com/stuff. Any mention of "w2lk" is incorrect. I have been informed that "wl2k" is the proper string. Honestly, if we're using a "Generic URL Template" than I believe that no validation whatsoever should be applied to the URL Scheme (defined as the characters to the left side of the colon in a URL).

closed 2025-02-10 by aanon4