aredn/aredn #1890: Missed version update in opkg distfeeds.conf URL

bwarden, 2025-02-11:
### Describe the bug
The opkg distribution feeds configuration in version 3.25.2.0 wasn't updated completely, so `opkg update` fails.
Base url includes releases/3/**24**/3.25.2.0 instead of releases/3/**25**/3.25.2.0`

- [x] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
`opkg update` should successfully update packages, and `opkg install ...` should successfully install packages

### Screenshots
```
# opkg update
Downloading http://downloads.arednmesh.org/releases/3/24/3.25.2.0/targets/ath79/mikrotik/packages/Packages.gz
*** Failed to download the package list from http://downloads.arednmesh.org/releases/3/24/3.25.2.0/targets/ath79/mikrotik/packages/Packages.gz

Downloading http://downloads.arednmesh.org/releases/3/24/3.25.2.0/packages/mips_24kc/base/Packages.gz
*** Failed to download the package list from http://downloads.arednmesh.org/releases/3/24/3.25.2.0/packages/mips_24kc/base/Packages.gz
...
```

### Additional context

aanon4, 2025-02-11:
Please try again.

bwarden, 2025-02-11:
Hah, I didn't even think about fixing it on the download server. Works, thanks!