aredn/aredn #1892: Some packages fail to reinstall after upgrade to Release 3.25.2.0

convolved-KY4G, 2025-02-13:
### Describe the bug
Some packages fail to reinstall automatically after upgrade to Release 3.25.2.0.  Some do successfully reinstall.  Incidentally, this likely also happened with nightly build 20250129-d446fe15 (r28417-daef29c75d), but I thought it was something I had done wrong.

### Expected behavior
Previously installed packages should reinstall upon a successful firmware upgrade.  The node had nine packages installed before the upgrade and only meshchat reinstalled automatically afterwards.  This is not what was expected.

### Support data

[supportdata-KY4G-GO-KIT-2025-2-13-17-48.tar.gz](https://github.com/user-attachments/files/18791425/supportdata-KY4G-GO-KIT-2025-2-13-17-48.tar.gz)

### Additional context
The meshchat package reinstalled fine.  The [https://github.com/kn6plv/WhereAndWhen](https://github.com/kn6plv/WhereAndWhen) package (v0.1.0) and its seven dependencies did not.  The dependencies are: gpsd, kmod-usb-acm, kmod-usb-serial, kmod-usb-serial-pl2023, kmod-usb-xhci-hcd, kmod-usb3, libusb-1.0-0.  The hardware was an AR300M16.

k1ky, 2025-02-15:
May not be related, but I've noticed that the IperfSpeed service listing pops back up after upgrades even though IperfSpeed module has been removed long ago.  This is still happening with 3.5.2.0 on nodes that used to have the module before but previously removed.

convolved-KY4G, 2025-02-22:
After I posted this, I noticed in the release notes that you must upgrade meshchat to v2.13 after upgrading to 3.25.2.0.  I still had v2.12 installed.  It is possible that not having that already upgraded could have affected the installation of the rest of the packages if it errored out somehow.  I will try re-flashing with 3.25.2.0 and report back what I find out.

convolved-KY4G, 2025-02-22:
So, my suspicion was correct.  Upgrading to 3.25.2.0 while MeshChat v2.12 was installed appears to have prevented the other packages from installing correctly after the attempted node auto-reinstall of MeshChat v2.12.  What is weird is it acted like the MeshChat v2.12 installation was successful; however it wouldn't run.  I do not know if MeshChat v2.13 is compatible with 3.24.10.0 and could be upgraded prior to v2.13 prior to the 3.25.2.0 firmware upgrade.  I am closing this ticket based on the findings.

closed 2025-02-22 by convolved-KY4G