aredn/aredn #1894: Port Forward Information is NOT being brought forward during firmware upgrades.

k1ky, 2025-02-15:
### Describe the bug
A clear and concise description of what the bug is.
Port Forwarding information does not appear to be transferred during upgrades.  This may be a long standing issue as I believe we have experienced this coming from several production versions.  I can confirm that from 3.24.10 and nightlies to 3.25.2.0 does not bring Port Forward information from devices.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes - 3.25.2.0

### Expected behavior
Expect proper migration of Port Forward information on devices from previous firmware.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
None available at this time.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
I can try to duplicate and provide support information if needed, but I don't think this information has been transferred for quite a while from previous versions.

k1ky, 2025-02-15:
> ### Describe the bug
> A clear and concise description of what the bug is. Port Forwarding information does not appear to be transferred during upgrades. This may be a long standing issue as I believe we have experienced this coming from several production versions. I can confirm that from 3.24.10 and recent nightlies to 3.25.2.0 does not bring Port Forward information from devices.  MikroTik HAP ac Lite.
> 
> * [ ]  Does this issue still occur on the latest version? (**Required**)
>   Yes - 3.25.2.0
> 
> ### Expected behavior
> Expect proper migration of Port Forward information on devices from previous firmware.
> 
> ### Support data
> Please add the support data for all nodes mentioned in this bug report (**Required**) None available at this time.
> 
> ### Screenshots
> Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
> None available as previous information is gone and the upgrades have no Port Forward information.
> If applicable, add screenshots to help explain your problem.
> 
> ### Additional context
> I can try to duplicate and provide support information if needed, but I don't think this information has been transferred for quite a while from previous versions if ever.  We rarely use Port Forward, but can be an issue with Yaesu DR (YSF) and Winlink Server operations.

aanon4, 2025-02-18:
Port forwarding was kept in multiple places in the old software, and the new software merges this all into once place. If you can provide a before upgrade / after upgrade support dumps showing how this fails I can try to work out what about this migration is failing. It usually works so I'll need a specific failure case to work with. Thanks.

nc8q, 2025-02-18:
[supportdata-NC8Q-Services1-2025-2-18-12-53.tar.gz](https://github.com/user-attachments/files/18855044/supportdata-NC8Q-Services1-2025-2-18-12-53.tar.gz)
[supportdata-NC8Q-Services1-2025-2-18-22-8.tar.gz](https://github.com/user-attachments/files/18855045/supportdata-NC8Q-Services1-2025-2-18-22-8.tar.gz)
One is 3.24.10.0 and the other is 3.25.2.0

aanon4, 2025-02-18:
https://github.com/aredn/aredn/pull/1910

k1ky, 2025-03-27:
Subsequent upgrades appear to be retaining port forward information, so I consider this taken care of with the possible awareness that migrating from older firmware before the fix may not bring forward the information.

closed 2025-03-27 by k1ky

AI7NC, 2025-05-11:
I can confirm my port forwards were dropped from 3.24.10 to 3.25.2.0 but have seemed to be retained beyond 3.25.2.0.