aredn/aredn #1896: Error Screenshot 3.25.2.0 Not Fatal

DIYsunnyday, 2025-02-17:
Error. 3.25.2.0 Screenshot attached. Not fatal. WiFi scan. Mikrotik hAP ac2. - W7VF

[image]

aanon4, 2025-02-17:
Does this happen the moment you open the wifi scan tool?

DIYsunnyday, 2025-02-17:
Hi Tim,

This is what I do:

Using Mesh 2.4 GHz, WAN Client 5 GHz.

1) In "Admin" mode, click "Node Tools".
2) Click "Wifi Scan".
3) Click "Radio: wlan0"
4) Select and click "wlan1"
5) Error display.
6) Press "Esc" on keyboard to clear.

Take Care,
Charles Whitlatch  W7VF

aanon4, 2025-02-18:
So this code has changed dramatically in the new nightly builds (for various unrelated reasons) and I cannot reproduce this bug there now. But if you were able, tomorrow after the next build is done, to try this again on that code I would appreciated it.
Thanks

aanon4, 2025-03-04:
No response after two weeks. Please reopen if necessary.

closed 2025-03-04 by aanon4

DIYsunnyday, 2025-03-05:
Thanks, Tim

You software-engineer so quickly we don't expect to see it again.

Charles Whitlatch  W7VF