aredn/aredn #1902: tx power on node status doesn't update when changed

BillHoover, 2025-02-18:
### Describe the bug
With the latest nightly I changed the tx power from 1dBm to 26dBm and applied the change. The radio info page did not reflect the change.
I also rebooted the node to see if that would cause the radio page to be updated, and it still did not.

- [Yes] Does this issue still occur on the latest version? (**Required**)
I do not know if it happened on versions previous to 20250217-001006c6

### Expected behavior
I would expect the radio page to agree with what was set for the RF link

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
[supportdata-AJ6UU-Pleasanton-1-2025-2-18-0-32.tar.gz](https://github.com/user-attachments/files/18834932/supportdata-AJ6UU-Pleasanton-1-2025-2-18-0-32.tar.gz)

### Screenshots

[image]

[image]

aanon4, 2025-02-18:
Well that's just odd. Looking at the support data the power is correctly set in all the places it should be, so quite how this reports as '1' I dont know. There are a few changes in next nightly build so please update and report back.

BillHoover, 2025-02-18:
I installed 20250218-d27965ce but nothing changed. I went through the radio settings page and changed the power down to 1dBm and then back up, committing the changes each time, but the node status page stays on 1. 
I don't know if it was clear before, but previously I *had* set it to 1 dBm, so that should have been correct before (although I actually find it hard to believe it actually was using that in the hardware - seems awfully weak).
I probably should also mention that when I tried to update the firmware today, the normal firmware dialog retrieved the correct updates, but I never could get it to actually load the nightly - it kept coming up with the red **ERROR: Unknown error** message. So I grabbed the update from the internet site and uploaded it. 
I didn't know what the mesh address it was trying to load was.
Unrelated, but I notice on my radio dialog that LQM is disabled. I *thought* that was on before, but I'm not sure. I also don't know if you think it would be better, in general, for me to enable that.

aanon4, 2025-02-18:
LQM is disabled during the testing of the new Babel code. LQM was helping out OLSR and I dont want it getting in the way of testing Babel. Leave it off for now.

Firmware download errors can be very transitory and due to many things, often related to other stuff going on in the network.

Could I have you set the password on your node to something you can temporarily share with me so I can take a more detailed look at it? If so just send me the password over Slack or email. Thanks.

k1ky, 2025-02-19:
I have seen this behavior, but I think it might be related to a browser update issue,  Does it show the "wrong" power level in a fresh load of the node in another browser?

aanon4, 2025-02-19:
https://github.com/aredn/aredn/pull/1912

closed 2025-02-20 by aanon4