aredn/aredn #1908: Mikrotik Hap AC Lite 3.25.2.0 Instability

kl5t, 2025-02-18:
### Describe the bug
Unit loses all tunnel connections.

- [ ] Does this issue still occur on the latest version? (**Required**) - Yes, it ONLY occurs on the current 3.25.2.0 version.  It did not occur on any previous versions.

### Expected behavior

Maintain tunnel connections

### Support data
Attached

[supportdata-KL5T-HAP-300-HOME-2025-2-18-10-52.tar.gz](https://github.com/user-attachments/files/18853811/supportdata-KL5T-HAP-300-HOME-2025-2-18-10-52.tar.gz)

Here are a number of lines from the end of the syslog:
Tue Feb 18 10:04:48 2025 kern.warn kernel: [70445.810420] 16384 pages RAM
Tue Feb 18 10:04:48 2025 kern.warn kernel: [70445.813582] 0 pages HighMem/MovableOnly
Tue Feb 18 10:04:48 2025 kern.warn kernel: [70445.824392] 1867 pages reserved
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.833921] Tasks state (memory values in pages):
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.845752] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.862283] [    380]    81   380      361       28    20480        0             0 ubusd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.884084] [    418]     0   418      290       10    20480        0             0 urngd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.899876] [   1057]   514  1057      525      156    20480        0             0 logd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.915505] [   1111]     0  1111      466       23    20480        0             0 rpcd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.937586] [   1344]     0  1344      303       11    16384        0             0 dropbear
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.953734] [   1512]     0  1512      971      241    24576        0             0 wpa_supplicant
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.976606] [   1679]     0  1679      337        8    20480        0             0 telnetd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70445.992557] [  16825]     0 16825      639      196    24576        0             0 netifd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.008311] [  16945]     0 16945      339       13    20480        0             0 vtund
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.023901] [  17520]     0 17520      341      107    20480        0             0 vtund
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.039661] [  17764]     0 17764      341      103    20480        0             0 vtund
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.055323] [  18029]   453 18029      418      218    20480        0             0 dnsmasq
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.077497] [  18212]     0 18212      341       96    20480        0             0 vtund
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.093498] [    715]     0   715     1049      213    20480        0             0 hostapd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.109337] [   2189]     0  2189      878      215    24576        0             0 uhttpd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.125177] [   3986]     0  3986      341       96    20480        0             0 vtund
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.134495] [   6553]     0  6553      341       96    20480        0             0 vtund
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.150224] [   8025]     0  8025      341      103    20480        0             0 vtund
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.166176] [   8906]     0  8906      870      207    24576        0             0 uhttpd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.188191] [   8908]     0  8908      308       11    20480        0             0 uclient-fetch
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.204814] [   8911]     0  8911      467       39    16384        0             0 modprobe
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.214428] [   8914]     0  8914      893      391    24576        0             0 uhttpd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.230275] [   8915]     0  8915      888      308    24576        0             0 uhttpd
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.246098] [   8917]     0  8917      299      112    20480        0             0 sh
Tue Feb 18 10:04:48 2025 kern.info kernel: [70446.261956] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=uhttpd,pid=8914,uid=0
Tue Feb 18 10:04:48 2025 kern.err kernel: [70446.284338] Out of memory: Killed process 8914 (uhttpd) total-vm:3572kB, anon-rss:900kB, file-rss:664kB, shmem-rss:0kB, UID:0 pgtables:24kB oom_score_adj:0

### Screenshots
Attached

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
The unit is GENERALLY accessible from the attached computer via the UI, but on one occasion it was not accessible from a browser and could only be accessed via WinSCP and Putty.

k1ky, 2025-02-18:
I concur as we are experiencing similar issues on HAP ac Lite Tunnel Server units 3.25.2.0

aanon4, 2025-02-18:
Your screen shot specifically notes it's running out of memory. Can we back off the number of tunnels and see how many it takes before it fails?

k1ky, 2025-02-18:
Here is a support data file where the gui doesn't load, but the node appears to be passing traffic to the internet and shows RF connections to other neighbors.  Am able to connect via WINSCP and run commands,

[supportdata.tar.gz](https://github.com/user-attachments/files/18855701/supportdata.tar.gz)

aanon4, 2025-02-18:
Running out of memory also. I suspect this is the return of dnsmasq eating all the memory for no good reason. We dealt with before release. I know what causes it, but now I need to actually fix it rather than just avoid it which is what we did for 3.25.2.0.

k1ky, 2025-02-18:
Glad to hear it.  I have a number of nodes that are doing this on our network.  Looking forward to the fix!

aanon4, 2025-02-19:
Just verifying that this is nightly only, yes?

k1ky, 2025-02-19:
No, Production 25.2.0 as well

kl5t, 2025-02-19:
No.  Current production.   3.25.2.0

aanon4, 2025-02-19:
@k1ky They strip emails out of message on here, but I will send it to you at your bell south address.

aanon4, 2025-02-19:
Here's my working theory.

dnsmasq is a pathological memory allocator/freer and the malloc/free implementation used in OpenWRT (from the library musl libc) is known to be poor. I'd perviously observed its performance to be very bad, but its ability to handle memory fragmentation is also known to be bad.

So, for this app I'm switching out the standard allocator for an alternative called "mimalloc" (https://microsoft.github.io/mimalloc/) which has much better performance. Let's see if this resolve this problem.

aanon4, 2025-02-19:
If you would like the test build to see if this resolve your issue please let me know where I can email it.

aanon4, 2025-02-19:
@kc2cbd I dont see the out of memory problems in your support data, or any problem I can quickly diagnose, so not this issue. Please open a separate issue with the details.

aanon4, 2025-02-20:
@k1ky feedback?

k1ky, 2025-02-20:
Still working with it - no issues so far, but I haven't put it on our real problem unit as of yet.

kl5t, 2025-02-20:
> Here's my working hypothesis.

Copying Tom Delker on this.....

Thanks,

Kent, KL5T

k1ky, 2025-02-20:
Got the test firmware installed on KL5T. Observing behavior - so far so good,

k1ky, 2025-02-20:
Should the Tunnel Server function be working with your HAP Test Firmware?

kl5t, 2025-02-20:
It looks like the custom version (KN6PLV-dnsmasq-node-setup-69480fd8) still has the issue.  It's losing connection to the outside as well.  At the moment I can still get to the unit via the UI at a locally connected computer via the unit's hotspot.  

I've attached the current support data file....

[supportdata-KL5T-HAP-300-HOME-2025-2-20-7-34.tar.gz](https://github.com/user-attachments/files/18891777/supportdata-KL5T-HAP-300-HOME-2025-2-20-7-34.tar.gz)

kl5t, 2025-02-20:
Here's a bit more.  Even though the unit reports a tunnel connection to ww4rfn-hap-200-jax (as indicated on the screen shots), I have no connections to any devices through that tunnel).  

And this is a TUNNEL - ONLY device.  There are no other AREDN units connected to it, other than tunnel connections.  It does have a Gandstream phone and a MeshPi connected, but that's it.

I've attached screen shots of the device's info screen, tunnel screen, and the "see what is on the local mesh" screen.

Strangely enough, a link to KI4QWT-PSM2-150-HSV appeared AND to WW4RFN-HAP-200-JAX appeared.  I can actually get to KI4QWT-PSM2-150-HSV but can't get to WW4RFN-HAP-200-JAX (I get a ww4rfn-hap-200-jax.local.mesh refused to connect).

[image]
[image]
[image]
[image]
[image]

k1ky, 2025-02-21:
2 of my servers are becoming unstable with the new test firmware.  Attempting to gain access and obtain support data.

aanon4, 2025-02-21:
Any update on this? Can you reach them via IP?

k1ky, 2025-02-21:
Still no go, I just tried IP.  You can see their tunnels answer k4pwo.mooo.com port 5525 and ww4rfn.hopto.org port 5525.

kl5t, 2025-02-22:
This morning the unit the UI is not accessible, however, the tunnel connections appear to be open.  I was able to telnet into the unit and generate a support file, attached here.  

[supportdata.tar.gz](https://github.com/user-attachments/files/18924003/supportdata.tar.gz)

aanon4, 2025-02-22:
Still running out of memory.
I rolled the memory change into nightly anyway as it seemed it did some good. Anyone running the custom build please upgrade to that and continue testing. Thanks.

k1ky, 2025-02-23:
I don't see a new nightly for 2/23 - will there be an update published today that includes this fix?

aanon4, 2025-02-23:
It was in 2/22

k1ky, 2025-02-24:
Nightly 0222 N3VUH support data (sluggish)

[supportdata.tar.gz](https://github.com/user-attachments/files/18934515/supportdata.tar.gz)

aanon4, 2025-02-24:
It's being hammered by incoming tunnel connections which are in turn being rejected. Probably not making it happy or responsive.

k1ky, 2025-02-24:
That seems to be the theme here - I'm trying my best to get those incoming tunnel units updated.  Will give it some time to stabilize and see what happens.  Can you tell why they are being rejected?

aanon4, 2025-02-24:
Lots and lots of "Auth failed" messages.

k1ky, 2025-02-24:
What would cause that? Any idea how to improve it?

k1ky, 2025-02-25:
Where are we on this? More to come?

aanon4, 2025-02-25:
The "Auth fails" are probably what they sound like - password are either wrong or the wrong node name is being used to connect. Might be good to eliminate this problem also. If this is because the tunnels are disabled, then best to re-enable them.
Also, switch to today's nightly and we'll see how that performs.

k1ky, 2025-02-25:
I am suspecting that the older 32MB tunnel devices are contributing to the issue. I'll pursue that line of testing and disable those connections to test my theory.

k1ky, 2025-02-28:
N0228 GUI refusing to connect.  Support file attached.

[supportdata.tar.gz](https://github.com/user-attachments/files/19033858/supportdata.tar.gz)

kl5t, 2025-02-28:
Just rebooted after working Tom K1KY above.  Here is a new support file after reboot:

[supportdata-KL5T-HAP-300-HOME-2025-2-28-13-40.tar.gz](https://github.com/user-attachments/files/19033885/supportdata-KL5T-HAP-300-HOME-2025-2-28-13-40.tar.gz)

aanon4, 2025-02-28:
@k1ky Unfortunately by the time you got to it, whatever the original problem was had long since been lost to the support data. I do notice you still have tunnels attempting, and failing, to connect. Can we fix those to the system is not constantly reconfiguring the network? Thanks.

aanon4, 2025-02-28:
@kl5t Your support file looks unremarkable. What were you seeing?

kl5t, 2025-02-28:
Do you have any ident on the tunnels that are attempting and failing to connect?  I'm not seeing them and thought we had them all disconnected.

kl5t, 2025-02-28:
The latest issue was that the Wifi hotspot was available and I was able to connect to it, but we could NOT get to the node via the GUI locally or via the tunnel connections (the tunnel connections were apparently open).  We WERE able to get to it via WinSCP.

Sometimes the Wifi hotspot even craps out and is unavailable.  Sometimes you can get to the GUI locally but can't get out to tunneled connections because they aren't open.  It's not always the same.

aanon4, 2025-03-01:
Quickly looking at the log file I can see KL7MM-SPENARD-HAP3-6HJ-172-31-148-76 coming and going. It's not the only one.

k1ky, 2025-03-01:
Yes, I see that - I'm working on updating all of the tunnel nodes to 0228, hopefully that will help.

K5RA, 2025-03-01:
I installed NB 20250228 in a hAP ac lite about 0630 this morning (Friday).  The node has 6 or 7 stable WG tunnels, and a DtD to a hAP ac2. By about 0735, the node had frozen as it had with 20250226 in previous days.  Even in the frozen state, ethernet comes through it to an RP5B, and I can ping the node's mesh and LAN addresses, but can not browse to it.  I have screen shots, and some Support Data, but not at the time node expired, only somewhat before.  
The hAP ac lite and the  hAP ac2 each have a WG tunnel to a remote hAP ac lite.  The tunnel between the two ac lites is usually the first to show failure.

k1ky, 2025-03-01:
K5RA: If you can ssh telnet login to the node you can generate a supportdata file by issuing the command: supportdata from the command line after logging into the node in character mode.

k1ky, 2025-03-01:
Support data N0228 HAP node is running low on memory, but GUI still currently available.

[supportdata-N3VUH-HAP-150-MBT-2025-2-28-22-27.tar.gz](https://github.com/user-attachments/files/19036892/supportdata-N3VUH-HAP-150-MBT-2025-2-28-22-27.tar.gz)

aanon4, 2025-03-01:
Still lots of tunnel churn.

k1ky, 2025-03-02:
N0228 HAP GUI still available but memory errors.  Should not be any stray tunnel attempts.

[supportdata-KL5T-HAP-300-HOME-2025-3-2-6-34.tar.gz](https://github.com/user-attachments/files/19043662/supportdata-KL5T-HAP-300-HOME-2025-3-2-6-34.tar.gz)

k1ky, 2025-03-04:
N0302 Supportdata file - HAP (non-tunnel) GUI doesn't load.  MeshChat is installed, but I have other MeshChat nodes with no problems.  Any idea what's hanging this one up?

[supportdata.tar.gz](https://github.com/user-attachments/files/19062030/supportdata.tar.gz)

aanon4, 2025-03-04:
Okay - today's build *should* address the memory issue we've been having. Please test and report back when you have a moment.

k1ky, 2025-03-05:
N204 Running out of memory low activity tunnel HAP

[supportdata-KM4EUG-HAP-150-MEDINA-2025-3-4-22-53.tar.gz](https://github.com/user-attachments/files/19082202/supportdata-KM4EUG-HAP-150-MEDINA-2025-3-4-22-53.tar.gz)

aanon4, 2025-03-05:
Never seen this one before in your reports here. Anything notable? Unfortunately the support data has been captured after the OOM evens have been going a while so I cant see what initially caused them.  It would be useful to catch it at that point.

aanon4, 2025-03-05:
Can you try the following:
1. Reboot the node
2. Telnet or ssh into the node
3. Type `/etc/init.d/zram disable`
4. Reboot

And then see how it does after that.

aanon4, 2025-03-06:
@k1ky Any feedback on those changes?

k1ky, 2025-03-06:
I've been applying 305 on a bunch of nodes. So far so good but the KM4EUG unit above that I applied the zram disable has become partially unresponsive and I can't get to it's gui nor ssh into it but it is passing traffic.

kl5t, 2025-03-06:
My node, KL5T-HAP-300-HOME is running 20250304-4a134d7e and this morning the GUI became unreachable.  The Wifi access point was still operating OK.  I've attached the supportdata file as well as the full text of what was available in the Syslog.  

[supportdata-KL5T-HAP-300-HOME-2025-3-6-7-59.tar.gz](https://github.com/user-attachments/files/19112593/supportdata-KL5T-HAP-300-HOME-2025-3-6-7-59.tar.gz)

[2025-03-07 0759 Syslog KL5T-HAP-300-HOME.txt](https://github.com/user-attachments/files/19112595/2025-03-07.0759.Syslog.KL5T-HAP-300-HOME.txt)

kl5t, 2025-03-08:
KL5T-HAP-300-HOME is still up and running since updating to 20250305-29beb78d 2 days ago.  I've attached current support data file and syslog entries.

I've noticed 6 occurences of an "oom-kill" followed by "Out of memory: Killed process" entries in the Syslog.  Don't know if this is of interest, just thought I'd point it out.  Please see below.

Regards,

Kent, KL5T

Sat Mar  8 01:50:40 2025 kern.info kernel: [29423.285526] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=manager.lua,pid=3876,uid=0
Sat Mar  8 01:50:40 2025 kern.err kernel: [29423.295476] Out of memory: Killed process 3876 (manager.lua) total-vm:3268kB, anon-rss:96kB, file-rss:520kB, shmem-rss:0kB, UID:0 pgtables:16kB oom_score_adj:0

Sat Mar  8 01:50:52 2025 kern.info kernel: [29435.950536] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=hostapd,pid=1565,uid=0
Sat Mar  8 01:50:52 2025 kern.err kernel: [29435.960289] Out of memory: Killed process 1565 (hostapd) total-vm:4060kB, anon-rss:156kB, file-rss:872kB, shmem-rss:0kB, UID:0 pgtables:16kB oom_score_adj:0

Sat Mar  8 02:07:54 2025 kern.info kernel: [30458.149714] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=manager.lua,pid=3922,uid=0
Sat Mar  8 02:07:54 2025 kern.err kernel: [30458.159690] Out of memory: Killed process 3922 (manager.lua) total-vm:3428kB, anon-rss:0kB, file-rss:180kB, shmem-rss:0kB, UID:0 pgtables:24kB oom_score_adj:0

Sat Mar  8 02:09:22 2025 kern.info kernel: [30546.399473] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=olsrd,pid=3726,uid=0
Sat Mar  8 02:09:22 2025 kern.err kernel: [30546.409090] Out of memory: Killed process 3726 (olsrd) total-vm:2844kB, anon-rss:12kB, file-rss:340kB, shmem-rss:0kB, UID:0 pgtables:20kB oom_score_adj:0

Sat Mar  8 02:11:44 2025 kern.info kernel: [30685.924817] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=hostapd,pid=3925,uid=0
Sat Mar  8 02:11:44 2025 kern.err kernel: [30685.934357] Out of memory: Killed process 3925 (hostapd) total-vm:4188kB, anon-rss:144kB, file-rss:464kB, shmem-rss:0kB, UID:0 pgtables:20kB oom_score_adj:0

Sat Mar  8 02:11:46 2025 kern.info kernel: [30693.996883] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),task=uhttpd,pid=3714,uid=0
Sat Mar  8 02:11:46 2025 kern.err kernel: [30694.006335] Out of memory: Killed process 3714 (uhttpd) total-vm:3868kB, anon-rss:0kB, file-rss:8kB, shmem-rss:0kB, UID:0 pgtables:24kB oom_score_adj:0

[supportdata-KL5T-HAP-300-HOME-2025-3-8-6-53.tar.gz](https://github.com/user-attachments/files/19144514/supportdata-KL5T-HAP-300-HOME-2025-3-8-6-53.tar.gz)

[2025-03-08 0648 Syslog KL5T-HAP-300-HOME FW20250305-29BEB78D.txt](https://github.com/user-attachments/files/19144518/2025-03-08.0648.Syslog.KL5T-HAP-300-HOME.FW20250305-29BEB78D.txt)

aanon4, 2025-03-10:
Looks like recent changes broke sysinfo.json's ability to correctly identify a mesh radio configuration. Thanks - I'll get that fixed.

kl5t, 2025-03-18:
Just updated to the most recent nightly (20250318) because the unit had become unreachable several days ago (I was out of town and couldn't deal with it).  After updating it today with the new nightly it didn't take too long to become unreachable via the GUI.  I was able to WinSCP and Putty into it and generated a support file...please see attached.

Thanks

Kent, KL5T

[supportdata.tar.gz](https://github.com/user-attachments/files/19329205/supportdata.tar.gz)

aanon4, 2025-03-18:
It ran out of memory. There isnt a memory leak, it just doesnt have enough memory to run everything active + babel during this transition from OLSR.

k1ky, 2025-03-18:
Is there anything we can remove to mitigate this shortcoming?  He needs to retain the Wi-Fi LAN Hotspot and it is a server connection for MeshChat (api) which shouldn't use much memory?

[8 more replies]