aredn/aredn #1914: Mesh PtMP not working in several ways

k6ccc, 2025-02-19:
### Describe the bug
I changed a hAP ac lite to Mesh PtP mode and the normally connected LHG-2nD-XL to Mesh station.  Afterwards, several problems observed.
1)  The Mesh PtMP node and the Mesh Station node will not connect.  If a WiFi scan is performed on the hAP, it does not see the LHG-2nD-XL.
2)  The hAP ac lite is not reachable via the mesh name nor the mesh IP.  It IS reachable via the LAN IP - either from the mesh or from a device connected to the LAP port.  I am currently unable to test this on the LHG-2nD-XL as the only connectivity I would have is via the mesh until I get home this evening.

DtD connections on the hAP ca lite DO show as connected, and apparently are functioning since the node is reachable via the mesh using the LAN IP.

- [ ] Does this issue still occur on the latest version? (**Required**)

All tests were done on 20250219-4574433a

### Expected behavior
The PtMP node and the Station node should connect.
The node(s) should be reachable via the mesh with the mesh name and mesh IP.

### Support data
Included for the hAP.  Unable to reach the LHG till I get home.

### Screenshots
I will add a couple in a few minutes.

[supportdata-K6CCC-hAP-Portable-2025-2-19-12-9.tar.gz](https://github.com/user-attachments/files/18875170/supportdata-K6CCC-hAP-Portable-2025-2-19-12-9.tar.gz)

k6ccc, 2025-02-19:
[image]
[image]

aanon4, 2025-02-19:
Looks like the channel on the radio is not set - possibly -2 isn't making it happy. Can you try with a non-negative channel? Thanks.

k6ccc, 2025-02-19:
Yes, it is on channel -2.  Has been that way for many months.  I can change the hAP end now, but unable to access the LHG until tonight after I get home.

aanon4, 2025-02-19:
Thanks. What channel it was on in some other mode doesnt really matter for this test. Let's try a non-negative channel when you are able.

k6ccc, 2025-02-19:
I have moved the hAP to channel 1.  I will change the LHG this evening.

aanon4, 2025-02-19:
Could you attach the support data? Thanks.

aanon4, 2025-02-19:
Let's keep test on the non-ham channels for the moment. There's probably more work to do to allow ham channels to work like this.

aanon4, 2025-02-19:
Yes

k6ccc, 2025-02-19:
Here you go.  And an update.  After an uptime of 1:22, I AM able to reach the hAP lite via both the mesh name and mesh IP.  When last I checked that was maybe an uptime of 15 minutes or so, and confirmed multiple times that only the LAN IP worked.

[supportdata-K6CCC-hAP-Portable-2025-2-19-14-31.tar.gz](https://github.com/user-attachments/files/18876473/supportdata-K6CCC-hAP-Portable-2025-2-19-14-31.tar.gz)

aanon4, 2025-02-19:
Thanks.
This time the management daemon (hostapd) came up because it thinks channel 1 is good (it thought -2 was bad) so the other end will have something to talk to once its on channel 1 also.

k6ccc, 2025-02-19:
We'll know in about 7 hours or so.

aanon4, 2025-02-20:
See also https://github.com/aredn/aredn/pull/1918 and https://github.com/aredn/aredn/issues/1917

k6ccc, 2025-02-20:
Home and connected to the LHG-2nD-XL. Changed it to channel 1, rebooted, rebooted the hAP lite (not at the same time). Refusing to connect. Here's support data for both nodes.

[supportdata-K6CCC-RB-LHG-2nD-XL-2025-2-19-21-49.tar.gz](https://github.com/user-attachments/files/18881801/supportdata-K6CCC-RB-LHG-2nD-XL-2025-2-19-21-49.tar.gz)
[supportdata-K6CCC-hAP-Portable-2025-2-19-22-18.tar.gz](https://github.com/user-attachments/files/18881802/supportdata-K6CCC-hAP-Portable-2025-2-19-22-18.tar.gz)

aanon4, 2025-02-20:
Can you try the following:
Login to the node:
Type /etc/init.d/wpad enable
Reboot

k6ccc, 2025-02-20:
I could execute those commands only on the hAP as I don't have any remote access to the LHG (I'm going to try to fix that tonight).  In any case, here is info on the hAP a few minutes after the reboot....

[image]

k6ccc, 2025-02-20:
And the Support Data

[supportdata-K6CCC-hAP-Portable-2025-2-20-6-32.tar.gz](https://github.com/user-attachments/files/18888062/supportdata-K6CCC-hAP-Portable-2025-2-20-6-32.tar.gz)

aanon4, 2025-02-20:
Thanks. It was the other end where this mattered unfortunately.

k6ccc, 2025-02-20:
That figures.  I will do that on the LHG-2nD-XL this evening.  I also have completed all the router / switch configuration to give me remote access to the LAN port on that node, but I need to run one 6 foot cable from the LHG to the family room LAN switch - and that I can't do remotely :)

k6ccc, 2025-02-21:
Did this on the LHG/2nD-XL:
Type /etc/init.d/wpad enable
Reboot

Link came right up.

[supportdata-K6CCC-RB-LHG-2nD-XL-2025-2-20-21-9.tar.gz](https://github.com/user-attachments/files/18901053/supportdata-K6CCC-RB-LHG-2nD-XL-2025-2-20-21-9.tar.gz)
[supportdata-K6CCC-hAP-Portable-2025-2-20-21-10.tar.gz](https://github.com/user-attachments/files/18901054/supportdata-K6CCC-hAP-Portable-2025-2-20-21-10.tar.gz)

aanon4, 2025-02-21:
Good stuff.

closed 2025-02-22 by aanon4