aredn/aredn #1916: Tunnel server display name is incorrect and links to different server

cameronzucker, 2025-02-19:
### Describe the bug
After upgrading to nightly build 20250219-4574433a and rebooting, a long-established tunnel server connection to the KD7DNY Phoenix Supernode now displays as k6gwe-tunnel-server. In details, the IP is still correctly pointing to KD7DNY-AZ-SUPERNODE (172.31.104.20:5535). After a sufficient amount of time has elapsed after reboot for the node to poll the Supernode connection, the Cloud icon will still link to KD7DNY-AZ-SUPERNODE, but clicking the neighborhood node link directly navigates to http://k6gwe-tunnel-server.local.mesh/

[image]

[image]

[image]

Because there is no actual connection to this server, the page fails to load or only partially loads, with differing levels of page load depending on how long the node has been booted with a connection to the internet. There has never been a link to k6gwe-tunnel-server saved on this node.

After the problem was first discovered, the node was rebooted and the [kd7dny-server-1](http://kd7dny-server-1.local.mesh/) tunnel server displayed correctly, but the issue is being reported in the event it's indicative of a deeper issue which needs investigation.

- [X] Does this issue still occur on the latest version? (**Required**)
Issue was observed after upgrading to 20250219-4574433a

### Expected behavior
The neighborhood node should display as [kd7dny-server-1](http://kd7dny-server-1.local.mesh/) and connect to KD7DNY-AZ-SUPERNODE's cloud node page. How it displays on a stable release node via DtD:

[image]

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-N7CPZ-M2PoEVM-2025-2-19-15-29.tar.gz](https://github.com/user-attachments/files/18876422/supportdata-N7CPZ-M2PoEVM-2025-2-19-15-29.tar.gz)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

Screenshots are provided in-line for context.

### Additional context
The connection to [kd7dny-server-1](http://kd7dny-server-1.local.mesh/) seems to be working in some respects but not others. It's polling the expected number of nodes through the network (462 OLSR nodes), but can't reliably reach the supernode page on clicking the Cloud icon.

aanon4, 2025-02-19:
Looks like we have two tunnel servers using the same set of tunnel addresses. This wouldnt matter as they're in different meshes, except for some reason someone has joined the Bay Area Mesh and the Phoenix Mesh together with a tunnel (not via supernodes) which is why you're seeing this problem.

Once that goes way (and I've not had chance to track down who's doing it - any ideas on your end?) this problem with resolve.

cameronzucker, 2025-02-20:
It's probably this connection: KJ6DLF-HAPAC to KJ6DLF-SAN-TAN-HAPAC2 TUN

Issue marked resolved. Thanks for investigating.

closed 2025-02-20 by cameronzucker

aanon4, 2025-02-20:
Connection between the meshes should be down. Hopefully things will look better now.