aredn/aredn #1917: Unable to find AREDN Mesh Station from PtMp device

Orv, 2025-02-20:
### Describe the bug
Both devices on channel 149, 10 MHz BW

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
PtMP station should see Mesh Station

### Support data

[supportdata-KB6DYJ-QTH-5G-2025-2-20-0-1.tar.gz](https://github.com/user-attachments/files/18879979/supportdata-KB6DYJ-QTH-5G-2025-2-20-0-1.tar.gz)
[supportdata-W6BI-TEST1-5G-2025-2-19-18-26.tar.gz](https://github.com/user-attachments/files/18879980/supportdata-W6BI-TEST1-5G-2025-2-19-18-26.tar.gz)

### Screenshots

### Additional context
I think I saw it once, but unsure.

aanon4, 2025-02-20:
Can you try this experiment.

Log into KB6DYJ-QTH-5G
Type `/etc/init.d/wpad enable`
Reboot

aanon4, 2025-02-20:
I think this is the fixed, but the above should confirm it.
https://github.com/aredn/aredn/pull/1918

Orv, 2025-02-20:
[supportdata-K6PVR-CAM-HILLS-5G-2025-2-19-21-8.tar.gz](https://github.com/user-attachments/files/18881125/supportdata-K6PVR-CAM-HILLS-5G-2025-2-19-21-8.tar.gz)
[supportdata-W6BI-TEST1-5G-2025-2-19-21-10.tar.gz](https://github.com/user-attachments/files/18881124/supportdata-W6BI-TEST1-5G-2025-2-19-21-10.tar.gz)

A fresh set of support data files.

aanon4, 2025-02-20:
Unfortunately I dont capture the allowed mac address in the support data, but the station is being rejected by the PtP node, which suggests its wrong in some way.

k6ccc, 2025-02-20:
Home and connected to the LHG-2nD-XL.  Changed it to channel 1, rebooted, rebooted the hAP lite (not at the same time).  Refusing to connect.  Here's support data for both nodes.

[supportdata-K6CCC-RB-LHG-2nD-XL-2025-2-19-21-49.tar.gz](https://github.com/user-attachments/files/18881778/supportdata-K6CCC-RB-LHG-2nD-XL-2025-2-19-21-49.tar.gz)
[supportdata-K6CCC-hAP-Portable-2025-2-19-22-18.tar.gz](https://github.com/user-attachments/files/18881777/supportdata-K6CCC-hAP-Portable-2025-2-19-22-18.tar.gz)

aanon4, 2025-02-20:
Can you try the following:
Login to the node:
Type `/etc/init.d/wpad enable`
Reboot

aanon4, 2025-02-20:
This is the part of the log where the station is trying to authenticate:
```
Wed Feb 19 21:05:54 2025 daemon.notice wpa_supplicant[1420]: wlan0: CTRL-EVENT-SSID-REENABLED id=0 ssid="AREDN-149-10-v3"
Wed Feb 19 21:05:54 2025 daemon.notice wpa_supplicant[1420]: wlan0: BSSID fe:45:58:ac:1d:68 ignore list count incremented to 2, ignoring for 10 seconds
Wed Feb 19 21:05:54 2025 daemon.notice wpa_supplicant[1420]: wlan0: Removed BSSID fe:45:58:ac:1d:68 from ignore list (clear)
Wed Feb 19 21:05:54 2025 daemon.notice wpa_supplicant[1420]: wlan0: SME: Trying to authenticate with fe:45:58:ac:1d:68 (SSID='AREDN-149-10-v3' freq=5745 MHz)
Wed Feb 19 21:05:54 2025 kern.info kernel: [  452.569248] wlan0: authenticate with fe:45:58:ac:1d:68 (local address=2c:c8:1b:ea:c3:5e)
Wed Feb 19 21:05:54 2025 kern.info kernel: [  452.578338] wlan0: send auth to fe:45:58:ac:1d:68 (try 1/3)
Wed Feb 19 21:05:54 2025 kern.info kernel: [  452.590065] wlan0: fe:45:58:ac:1d:68 denied authentication (status 1)
Wed Feb 19 21:05:54 2025 daemon.notice wpa_supplicant[1420]: wlan0: CTRL-EVENT-AUTH-REJECT fe:45:58:ac:1d:68 auth_type=0 auth_transaction=2 status_code=1
Wed Feb 19 21:05:54 2025 daemon.notice wpa_supplicant[1420]: wlan0: Added BSSID fe:45:58:ac:1d:68 into ignore list, ignoring for 10 seconds
Wed Feb 19 21:05:54 2025 daemon.notice wpa_supplicant[1420]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="AREDN-149-10-v3" auth_failures=3 duration=30 reason=CONN_FAILED
```

aanon4, 2025-02-20:
I did find the mac address you set in a different part of the configuration, and it is correct. But, as I said earlier, I dont log the actual place the information is stored for the radio to use (that'll be fixed tomorrow) so maybe I mess that up somehow.

Orv, 2025-02-20:
Testing 20250220.  Mesh station now sees PtP station, but not the other way.

[supportdata-K6PVR-CAM-HILLS-5G-2025-2-20-10-53.tar.gz](https://github.com/user-attachments/files/18893450/supportdata-K6PVR-CAM-HILLS-5G-2025-2-20-10-53.tar.gz)
[supportdata-W6BI-TEST1-5G-2025-2-20-10-54.tar.gz](https://github.com/user-attachments/files/18893451/supportdata-W6BI-TEST1-5G-2025-2-20-10-54.tar.gz)

aanon4, 2025-02-20:
Odd. The authorization file on W6BI-TEST1-5G is empty.
Could you log into the above node, type `/etc/init.d/manager restart`, and see if anything changes (give it a few minutes). Then send the support data for this node again, whether it connects or not.

aanon4, 2025-02-20:
For clarification - did you do the instructions above or did you restart the node?

aanon4, 2025-02-20:
Okay. Looks like it connected. The last log entry shows it connected. How long did you give it after that?

Orv, 2025-02-20:
I did the restart command you gave.   It was a couple of minutes.  After the failed attempt to connect to the far end, I ran that restart command again, but this time it didn't appear to connect.

aanon4, 2025-02-20:
Okay. Can we step back to PtMP mode (no PtP) and verify everything there is working?

Orv, 2025-02-20:
PtMP <--> Mesh Station works as expected.   100/100 LQ/NLQ on both ends, SNR ~ 70 dB.  Iperf3 shows 15-19 Mbit/sec in both directions.

aanon4, 2025-02-20:
Okay. Let's stop there until tomorrow's build.

Orv, 2025-02-21:
Testing 20250221.  PtP <--> Mesh Station now works as expected.    Tested on channel 149 & 178.   iperf3 shows noticeably improvement in throughput (but more thorough testing needed).

Orv, 2025-02-21:
OK to close.

closed 2025-02-21 by Orv