aredn/aredn #1928: Supernode loss of external connectivity

Orv, 2025-02-22:
### Describe the bug
Supernode is accessible via DHCP, but not pingable, nor are DtD links operational

- [ ] Does this issue still occur on the latest version? (**Required**)
No
### Expected behavior
Should function as normal

### Support data

[supportdata-W6BI-SOCAL-SUPERNODE-2025-2-22-9-38.tar.gz](https://github.com/user-attachments/files/18924512/supportdata-W6BI-SOCAL-SUPERNODE-2025-2-22-9-38.tar.gz)

### Screenshots

[image]

aanon4, 2025-02-22:
Running out of memory. Reboot it and stick today's night on it.

Orv, 2025-02-22:
Applied latest nightly build and rebooted.   Functionality restored.

closed 2025-02-22 by Orv