aredn/aredn #1929: Node doesn't propagate service DNS entries properly with LAN in NAT mode

cameronzucker, 2025-02-23:
### Describe the bug
Node LAN mode was changed to NAT to accommodate a larger IP range than 29 clients. After the change was made, DHCP reservations were updated, DNS records re-associated with hostnames, and port forwarding rules put in place for the following ports:

80, 443, 8080, 2222, 23, 53, 123 forwarded 1:1 - localnode

Other ports were forwarded as required, either 1:1 to their hosts (pvem3 - 8006) or with asymmetric forwarding rules to handle multiple hosts/services listening on the same port (wiki - 81:80).

Example screenshots:

[image]

[image]

In Network -> Advanced options, LAN default route was enabled.

Port forwarding rules were tested to work from a DtD node's LAN using the NAT node's mesh IP:port, including for asymmetric port mappings:

[image]

However, DNS fails to propagate for the new entries, both to other nodes locally and to the supernode network. After the node reboots and several minutes have passed, it shows up on the supernode network with only localnode associated services:

[image]

nslookup of DNS aliases fails from a DtD node's LAN:

[image]

Test-NetConnection fails in a similar fashion:

[image]

- [X] Does this issue still occur on the latest version? (**Required**)
Confirmed in 3.25.2.0 and 20250222-2237fdee
### Expected behavior
Local service entries and DNS records should propagate through the mesh using both DtD and Supernode connections.

### Support data
N7CPZ-HalfRackVM data added as well to show what a DtD node sees. N7CPZ-M2PoEVM is in NAT mode.

[supportdata-N7CPZ-M2PoEVM-2025-2-22-22-18.tar.gz](https://github.com/user-attachments/files/18928238/supportdata-N7CPZ-M2PoEVM-2025-2-22-22-18.tar.gz)

[supportdata-N7CPZ-HalfRackVM-2025-2-22-22-18.tar.gz](https://github.com/user-attachments/files/18928239/supportdata-N7CPZ-HalfRackVM-2025-2-22-22-18.tar.gz)

### Screenshots
Screenshots included in-line for context.

### Additional context
Tried several variations on port forwarding and default route enabled/disabled to see if DNS could be made to propagate (like adding port 53 TCP/UDP to localnode), but none of these changes made any difference. Hostname lookup succeeds locally if not using the advertised service reservation, so hosts can be reached via http://hostname.local.mesh:port, but not via http://dnsalias.local.mesh:forwardedport.

aanon4, 2025-02-24:
The current behavior in NAT mode is that we dont publish any aliases because they aliases to addresses which cannot be reached from the mesh (ie. they're behind the NAT). Would it make more sense to publish these names with the node's address instead, so they will resolve and the NAT will do its thing? I'm not a user of NAT mode so I dont have a good feel for what makes sense.

cameronzucker, 2025-02-24:
We worked on this for a few hours tonight testing and considering possible solutions. "Would it make more sense to publish these names with the node's address instead" - yes, exactly this: have the host DNS aliases point at the node hostname for external mesh connections.  It doesn't really matter if they're A records pointing to the mesh IP or CNAME records which point to the node name. Either is fine, and I'm not sure which we're already using.

In summary, because many services only work well with a single "configured URL (e.g., call-example.local.mesh), it's ideal to have be able to have a consistent service alias that works on both mesh and LAN networks.
Split DNS in NAT mode achieves this. You would have:

Mesh DNS: (desired new behavior)
(A) call-example.local.mesh -> $MeshNodeIP, or
(CNAME) call-example.local.mesh -> $MeshNodeName (-> $MeshNodeIP)

Internal DNS: (Current "Local Services" local DNS CNAME mapping behavior)
(CNAME) call-example.local.mesh -> $TargetHost

aanon4, 2025-02-25:
Okay https://github.com/aredn/aredn/pull/1933 is taking a stab at advertising hostname which are NATed.

cameronzucker, 2025-02-25:
Which ports do nodes use to communicate information to the supernode network?

Setting something other than the node to use 82 caused this:

[image]

aanon4, 2025-02-25:
Sorry not sure what you mean by the above. If you are verifying this issue please let me know what has changed, what works, what doesn't, and include support data. Thanks.

cameronzucker, 2025-02-26:
The node normally named N7CPZ-M2PoeVM is now advertising itself as N7CPZ-Certs.local.mesh, both on DtD and the supernode network.

[supportdata-N7CPZ-M2PoEVM-2025-2-25-17-41.tar.gz](https://github.com/user-attachments/files/18974741/supportdata-N7CPZ-M2PoEVM-2025-2-25-17-41.tar.gz)

aanon4, 2025-02-26:
And can I assume that N7CPZ-Certs is a nat-ed host on N7CPZ-M2PoeVM?

cameronzucker, 2025-02-26:
As far as the node is aware, N7CPZ-Certs exists as the following:

Advertised service:
[image]

DNS host alias:
[image]

With these port forwarding rules:
[image]

It should be pointing an an nginx web server VM which has wildcard cert details on a static webpage: http://n7cpz-certs.local.mesh:83/

I thought there may have been an existing use with port 82 when I first saw this issue with how the node was advertising itself so I shifted the service and all associated entries to 83 with no change.

This is what the node thinks it should look like on the local mesh:
[image]

I'm also attaching support data from a DtD node for your reference, so you can see what it sees:

[supportdata-N7CPZ-HalfRackVM-2025-2-25-18-0.tar.gz](https://github.com/user-attachments/files/18974927/supportdata-N7CPZ-HalfRackVM-2025-2-25-18-0.tar.gz)

cameronzucker, 2025-02-26:
Just wanted to add: the feature we set out to achieve works. For example, http://n7cpz-pvem3.local.mesh:8006, with 8006:8006 port forwarding, works from outside of LAN on the mesh. Just seems like there were some side effects.

After a couple reboots the node name on DtD went back to normal, but it still shows as certs.local.mesh on the Supernode network.

aanon4, 2025-02-27:
The way OLSR distributes its name/ip mappings isnt like DNS unfortunately, so we dont get CNAME or anything useful like that - strictly name-to-ip pairs. To the reverse lookups are as likely to give one of the various aliases as anything else.
I can maybe mitigate this, but I need an example from you exactly where the wrong name is being show. Thanks.

cameronzucker, 2025-02-27:
Here's how the node currently shows running firmware 20250226-36475386:

[image]

DtD link stopped working sometime yesterday. I've seen it cycle between a few services - most recently, it showed up on both the local DtD link and supernode network as n7cpz-tiles.local.mesh, with the following associated service, ports, and DNS records:

[image]

[image]

[image]

Behind NAT, the node's Apache server also seems to try intercepting and serving requests on port 80. For example http://n7cpz-wiki.local.mesh is a port 80 service. When I go to that URL, which is DNS mapped to mobilewiki.local.mesh, I should get this:

[image]

But what I actually get is this:

[image]

The hostname resolves, but it's trying to load the /a/status directory from the node's Apache server. Let me know if you need any other supporting materials. I can probably set up a tunnel for you straight to the M2PoEVM node if that would help. I've already got DDnS setup on the router.

cameronzucker, 2025-03-17:
We can abandon this issue and mark "functioning as designed."

closed 2025-03-17 by cameronzucker

aanon4, 2025-03-17:
So I should remove https://github.com/aredn/aredn/pull/1933 ?

cameronzucker, 2025-03-17:
Yes, as things stand I just don't think it's compatible with OLSR. We can potentially revisit after the full switchover to Babel down the road.

aanon4, 2025-03-17:
https://github.com/aredn/aredn/pull/1995