aredn/aredn #1930: Various Babel errors found in logs

Orv, 2025-02-23:
### Describe the bug
Various babel errors in the log

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
Wouldn't expect to see these errors under normal conditions

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

[supportdata-W6BI-SOCAL-SUPERNODE-2025-2-23-15-40.tar.gz](https://github.com/user-attachments/files/18934388/supportdata-W6BI-SOCAL-SUPERNODE-2025-2-23-15-40.tar.gz)

aanon4, 2025-02-23:
It's running out of memory. This seems to be a persistent issue with dnsmasq at the moment when it's forced to load and reload as the DNS entries change. Why this is more of an issue now than in previous openwrt release is not clear.

Orv, 2025-02-23:
:-/

Orv, 2025-02-23:
Here's what my AI came up with:

    Each reload of dnsmasq clears its cache and re-reads configuration files. If this happens too frequently (e.g., due to constant SIGHUP signals or service restarts), it can cause high memory usage as the cache is repeatedly rebuilt.

    Configuration options like clear-on-reload may exacerbate this by forcing the cache to be flushed on every reload.

    If the system running dnsmasq has limited memory, frequent reloads can overwhelm available resources. This is especially common in constrained environments like routers or small embedded systems.

    High memory usage can also occur if dnsmasq is caching a large number of DNS queries or handling a large volume of DHCP leases.

    Misconfigured options, such as enabling unnecessary features or using overly large cache sizes (cache-size), can lead to excessive memory consumption.

    Using the no-poll option incorrectly might result in unexpected behavior when configuration files like /etc/resolv.conf are updated.

    External processes (e.g., DHCP lease updates, changes to /etc/hosts, or automated scripts) may trigger frequent reloads unnecessarily. This can result from poorly implemented automation or monitoring tools that send SIGHUP signals too often.

    For example, tools like Chef or Ansible might trigger multiple reloads during configuration management if notifications are not properly consolidated.

    If dnsmasq cannot write to its lease file (/var/lib/misc/dnsmasq.leases) due to insufficient disk space, it may fail and consume additional memory while retrying operations.

    A high volume of DNS queries or malformed requests can overload dnsmasq, leading to increased CPU and memory usage. This is common in environments with misbehaving clients or malicious traffic.

    Consolidate configuration changes and ensure only a single restart/reload occurs after all changes are applied (e.g., use delayed notifications in tools like Chef)

    Use the --hostsdir option instead of SIGHUP for dynamic updates to /etc/hosts, as it avoids reloading other configurations unnecessarily.

    Limit the cache size (cache-size) to an appropriate value for your environment.

    Avoid enabling unnecessary features like dynamic DHCP lease registration if not needed.

    Check system logs for memory-related errors and review CPU/memory usage patterns.

    Ensure sufficient system resources (RAM and disk space) are available.

    Investigate what is triggering frequent reloads (e.g., DHCP lease changes, external scripts) and address the root cause.

    Increase system resource limits for dnsmasq if possible, especially on constrained devices.

aanon4, 2025-02-24:
Okay - let's switch the babel dnsmasq mechanism over to how we do it for olsr. I'd rather not, but this seems to use less memory.

Orv, 2025-02-24:
Updated to latest nightly.  What should I look for in the logs, or?

aanon4, 2025-02-24:
Look for suspiciously looking backtraces in the syslogs with the phrase "oom"

aanon4, 2025-02-25:
You're supernode is still running out of memory :-(

aanon4, 2025-03-04:
Hopefully fixed now the memory issues are fixed.

closed 2025-03-13 by aanon4