aredn/aredn #1935: AREDN with 44 IPv4 WAN-Adresses

RRERF, 2025-02-25:
### Describe the bug
The "HamNet" has been in operation in Europe for around 15 years. This is a high-altitude network that stretches across the whole of Europe.
 
We use this network to network the AREDN nodes. This means that we establish VPN connections via this HamNet. The routers in the HamNet are in the 44 IP address range.

- [X] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Since the introduction of the 44 function in AREDN, the phenomenon has arisen that the VPN connection is established and as soon as communication is possible, the VPN connection breaks off because the 44 address (which is actually on the WAN of the AREDN router) is searched for via the VPN (i.e. LAN).

### Support data
[HamNet-Map Europe](https://hamnetdb.net/map.cgi)
[HamNet Informations](https://de.wikipedia.org/wiki/HAMNET)
[Part of HamNet](https://www.hb9cf.ch/index.php/ham-net/)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
My Callsign is HB9FRR, Name Roman, i also work together with HB9FTS, Alex.

[image]

[image]

aanon4, 2025-02-25:
For the moment I suggest you disable 44net on your local supernode as this is probably causing your problems.
https://docs.arednmesh.org/en/latest/arednHow-toGuides/supernodes.html#net-and-supernodes

RRERF, 2025-02-26:
Hello
Thank you for this Information.
Basically, our system works again if we deactivate the 44Net option on the supernode.
Unfortunately, we have VPN connections to other "regions" (HB9FTS, etc.) over the Internet on various nodes. This means that I had to deactivate all connections to other supernodes and VPN servers as well. So we are working in "isolated operation".

I suggest setting the routing metric of the local addresses (in our case, 44.X.X.X on the WAN interface) lower than the subnets that are reached via the VPN. This is state of the art in network technology.

aanon4, 2025-02-27:
Thanks for this. Could you give me some examples of your exact IP/subnet for your VPNs? Out 44.X routes should already be lower priority, but clearly this isn't working as intended.

RRERF, 2025-03-03:
Hello
So, in Europe, the 44subnets are splitted to regions and cities:
[HamNet-DB IP-Adresses](https://hamnetdb.net/?m=subnet&showAllLines=1)

Examples:
Switzerland: 44.142.0.0/16
Austria: 44.143.0.0/16
Germany: 44.148.0.0/16
Netherland: 44.137.0.0/16

So, we use the existing HamNet, to connect AREDN-VPNs, because there are a lot of ham-operators, which don't want to change the existing hamnet, and, the hamnet is existing since 15years.

aanon4, 2025-03-03:
The goal is for AREDN and HamNet (and other uses of 44-net) to play nice together. Clearly not there yet. I appreciate the information so we can make it better.

RRERF, 2025-03-03:
This would be great!
At the Moment, our AREDN-Net is working just through the Internet and the 2 Links we have with AREDN. So it's not really "Emergency-Radio"...