aredn/aredn #1945: NB 20250302 Prevents WG Tunnels

K5RA, 2025-03-03:
System is hAP ac lite with RPi5B support computer with Chromium browser.  

[supportdata-K5RA-HAP-FB-2025-3-3-6-52.tar.gz](https://github.com/user-attachments/files/19054496/supportdata-K5RA-HAP-FB-2025-3-3-6-52.tar.gz)
[image]
[supportdata-K5RA-HAP-FB-2025-3-2-20-41.tar.gz](https://github.com/user-attachments/files/19054497/supportdata-K5RA-HAP-FB-2025-3-2-20-41.tar.gz)
[image]

After installing NB 20250302 in hAP ac lite, one of four tunnel clients failed to go active and three of three tunnel servers failed to go active.  Multiple reboots and disabling/enabling tunnels had no effect. 
When looking the node with ssh, I saw on order of fifteen seconds delay between typing commands (e.g., ls -l) and response appearing on command line, then similar delays before command results after pressing Enter.  In one case, something about "broken pipe" appeared on screen, then delays disappeared.  
The browser displays often exhibited similar delays.

After re-installing release 3.25.20.0, all seven WG tunnels were active.  

Expected performance is that all node capabilities will work with nightly build.

Before and after screenshots and support data provided.

K5RA, 2025-03-03:
Check out the LAN Hotspot and WiFi Client entries in before and after screen shots.   The setups were in the node before re-installing to 3.25.2.0, but NB 20250302 would not display them.  All I did was install the stable release, and let the node come up.

K5RA, 2025-03-04:
I have switched my K5RA-HAP-FB node from NB 20250302 to release 3.25.2.0 to NB 20250304 to release 3.25.2.0.  When node has the NB, the three WG tunnels served by that node go inactive, and the Wifi WAN and LAN setups are nonsense.  With the stable release, all available tunnels are active (two nodes are at a location with no Ethernet today) and the WAN and LAN setups become what they should be. 

I have plenty of screenshots and support data for all cases, but I believe results are same as two cases above.

K5RA, 2025-03-05:
Uploading NB 20250305 to replace 3.25.2.0 did same thing.  Three tunnel servers went inactive.  Wifi WAN and LAN setups were nonsense.  Screenshots and support data available.

K5RA, 2025-03-07:
Ditto NB 20250307.  I did a brief test, took some data, then reverted back to 3.25.2.0.  Node unused memory is about 15% with both NB an stable release.

aanon4, 2025-03-13:
As there have been many changes since this original report, and there will be more tomorrow, can you do the following:

1. Update to nightly 20250313
2. Setup the node such that the tunnels which fail are failing.
3. Create new support data and attach it here.

Thanks

K5RA, 2025-03-13:
Thank you for stepping in here.  
I installed NB 20250313 over 3.25.2.0 in my hAP ac lite (HAP-FB).  Same results as before:  the WAN Client went away, and the tunnel servers went inactive.  In a separate experiment, I found that turning off the WAN Client when running 3.25.2.0 inactivated the Tunnel Servers.  
Configurations as follows.
HAP-FB is a hAP ac lite that has wired Ethernet from ISP on Port 1.  It is configured with WAN Client on the 2.4 GHz radio with ISP WiFI SSID.  It has HAP2 (below) as DtD (Port 5) and an RPi5 service computer (Port 2).  HAP-FB serves out four WG tunnels.  One remote node is not active, so tunnel active/allocated count shows 3/4 when tunnel servers are active.  
HAP2 is a hap ac2, co-located with HAP-FB, that has wired Ethernet from ISP on Port 1.  It is running NB 0310.  It is configured with 2.4 GHz radio off, and 5GHz radio on ISP WiFi SSID, but node Status page shows "no connection."  Ports 3, 4, and 5 are configured as DtD with the HAP-TC (hAP ac lite Tower version) RK2C (Rocket), and HAP-FB attached.  
HAP-WH is a remote hAP ac lite.  It serves a WG tunnel to HAP-FB.  It is running NB 0310.  It has wired Ethernet on Port 1.  It has a LAN hotspot on the 5 GHz radio and a WAN Client on the 5GHz radio.  
Here is HAP-FB Status page screen shot in stable release configuration.

[image]

And here is the HAP-FB support data in stable release configuration.

[supportdata-K5RA-HAP-FB-2025-3-13-9-56.tar.gz](https://github.com/user-attachments/files/19231967/supportdata-K5RA-HAP-FB-2025-3-13-9-56.tar.gz)

Here is the HAP-FB Status page screen shot after NB 20250313 was installed. 

[image]

And here is the HAP-FB support data after NB 20250313 was installed.

[supportdata-K5RA-HAP-FB-2025-3-13-10-11.tar.gz](https://github.com/user-attachments/files/19232019/supportdata-K5RA-HAP-FB-2025-3-13-10-11.tar.gz)

Here is the HAP2 screen shot after NB installed in HAP-FB.  

[image]

Here is the HAP-WH screen shot after NB installed in HAP-FB. 

[image]

I have support data from HAP2 and HAP-WH, but figured there was already TMI.

K5RA, 2025-03-13:
The simple answer is "I don't know."  HAP-FB and HAP-WH were installed first.  I decided to try a hAP ac2 (it had more memory so might work better with Meshchat), and configured it like the others.  Originally, all three had both 2.4 and 5 GHz radios in use as WAN Clients and LAN Hotspots.  The WiFi links seemed not work as we wanted, so I replaced them with CAT5 cable, but did not delete the WiFi setups.  The WiFi setup on HAP-FB was changed recently (several times) trying to figure out what was happening to Tunnel Servers.  

But now, when HAP-FB has no WiFi (both radios OFF), the Tunnel Servers go inactive, but the Tunnel Clients stay up.  Maybe that WHY? is the question to answer.

aanon4, 2025-03-13:
So one problem here is I dont know how you got into this state, and moving backwards and forwards between release and nighty has muddied things a fair bit. To explain a bit; we're migrating how the wireless state is managed internally. The old way was very restrictive in what we could do, while the new way will let us do more in the future. When you first migrate to nightly from release we transform the state from one to the other. However, we dont delete the old state because that would completely break reverting the firmware. But, if you revert and then install nightly again, because the state has been converted once, it wont be converted again. This is the point where the two configurations - release and nightly - can get out of sync as changes in one will no longer propagate to the other.

So right now, running nightly, the wan wifi client is off, and its off because new configuration says it should be off. Because of this, the hap is using the wired WAN connection and, you'll note, that this has a different IP address to the one you had when using the WiFi WAN. That's why your outgoing tunnels work - they dont care what your IP address is - but your incoming ones dont - your firewall isnt forwarding to the correct IP address now because it changed.

Next thing to do is to turn your Wifi Client WAN back on like it was when running release. This should now be possible because of other fixes that happened in nightly recently.  Give that a try.

K5RA, 2025-03-13:
Rcvd.  Thanks.  It has been a mystery to me why HAP-FB was different.  I got hAP-TC to be a replacement for it, but was not sure if installing the  Backup from HAP-FB into HAP-TC would work at all, or just bring over the same bad stuff.  

I have seen that when I go from stable to NB to stable, the WiFi setups are restored.  

I need to do about  90 mins of PT, then will put NB 0313 into HAP-FB and try to set up a WiFi Client on one of the radios.  That has not been successful when tried with some older NBs.  Back to you in couple of hours.

K5RA, 2025-03-13:
The HAP-FB node started with NB 20250313 active, and Tunnel Servers not active.  Screenshot

[image]

Here is the support data for that setup. 

[supportdata-K5RA-HAP-FB-2025-3-13-13-33.tar.gz](https://github.com/user-attachments/files/19235255/supportdata-K5RA-HAP-FB-2025-3-13-13-33.tar.gz)

I set up the 2.4 GHz radio to be a WiFi Client on the ISP router.  When the node came back from reboot, the tunnel servers were active!

[image]

Here is the WiFi set up screen.

[image]

And here is the support data for that. 

[supportdata-K5RA-HAP-FB-2025-3-13-13-39.tar.gz](https://github.com/user-attachments/files/19235289/supportdata-K5RA-HAP-FB-2025-3-13-13-39.tar.gz)

Then I deleted the 2.4 GHz radio setup, and set the 5GHz radio as the WiFi Client. 

[image]

And that disabled the Tunnel Servers. 

[image]

Here is the support data for that.

[supportdata-K5RA-HAP-FB-2025-3-13-13-46.tar.gz](https://github.com/user-attachments/files/19235320/supportdata-K5RA-HAP-FB-2025-3-13-13-46.tar.gz)

I hope this helps.  I am glad to have a configuration that works with the NBs.  Thank you very much.

aanon4, 2025-03-13:
Closing this now.

Summary:

When the wifi wan client got disabled, the wire wan is used which means a different IP address for the device, which means you firewall is not forwarding to this new IP, and incoming tunnels fail.

closed 2025-03-13 by aanon4