aredn/aredn #1948: 3.25.2.0 on PowerBeam M2 does not permit LAN netwoirk choice in NAT mode

ka2ddo, 2025-03-04:
### Describe the bug
On a Ubiquiti  PowerBeam M2-400, freshly flashed to 3.25.2.0 from factory code, I am unable to choose the IPv4 network number for my LAN when the node is in NAT mode. No matter how many times I try to change the network number, it snaps back to 172.27.0.0/24.

- [YES] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
I expected to be able to set the LAN network number to 192.168.0.0/24 (the network my servers are running on) so I could open port forwarding from the mesh to my local servers.

### Support data
[supportdata-KA2DDO-PBM2-2025-3-3-19-46.tar.gz](https://github.com/user-attachments/files/19062235/supportdata-KA2DDO-PBM2-2025-3-3-19-46.tar.gz)

[image]

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
My previous node (before hardware failure) at a older minor release permitted me to choose the network number for my NATted LAN. Using an additional router doesn't work because the node does not permit hops past the LAN network to another network beyond it.

aanon4, 2025-03-04:
Could you also add a screenshot for your attempted values before hitting the Done button? Thanks.

ka2ddo, 2025-03-04:
Here's the screenshot as I want it to be before clicking Done.

[image]

aanon4, 2025-03-04:
I can't see under the dialog, but can you check that LAN DHCP is enabled? Dialog middle/right of the screen.

ka2ddo, 2025-03-04:
I specifically have DHCP turned off, because I already have a DHCP server on the 192.168.0.0/24 LAN (and all the servers on that LAN use static addresses, not DHCP addresses, so they will stay put for port forwarding purposes).

aanon4, 2025-03-04:
Understood. Could you turn it on to see if you can change the values like that, and that they persist once you turn if off again?

ka2ddo, 2025-03-04:
That was kludgy, but it worked. I now have the network number I want for my LAN, and DHCP turned back off before it fights with my other DHCP server and screws up the kids' school laptops to tell them the school LAN (over VPN) is on the mesh.

aanon4, 2025-03-04:
Kludgy and a bug that needs to be fixed. Thanks for reporting it.

aanon4, 2025-03-04:
https://github.com/aredn/aredn/pull/1950