aredn/aredn #1961: http server unreacchable

kp4djt, 2025-03-08:
### Describe the bug
After a period of time I have not been able to determine but at least 2 hours attempts to reach the node web page end in Site unreachable. Prior to the site going to sleep the web page is slow to respond. But the node beyond it responds quite rapidly so the network path through the node is good.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
Attempt to reach site web page should produce said page 

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)
On attempt to pull support data node locked up and died, tried 3 times twice from web screen and once from command line. All attempts resulted in machine doing what appears to be a reboot

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-03-08:
Thanks for creating this; but without support data or a way to reproduce this, I cant do much.

aanon4, 2025-03-08:
The node is `N5KNU-LB5-QTH-2-ESN` (not stated above)

kp4djt, 2025-03-08:
I will continue to try to get the support data. Right now when I checked
it, it was in that funny state where
it shows sort of a partial connection but will not resolve. Not sure how
long it has been that way. As soon
as I can get into it I will turn off the hotspot.

aanon4, 2025-03-08:
I've disabled the hot spot, which will stop all the WPA features from running as they take a fair bit of memory. Let's see if that stabilizes things.

aanon4, 2025-03-08:
[supportdata-N5KNU-LB5-QTH-2-ESN-2025-3-8-10-58.tar.zip](https://github.com/user-attachments/files/19144833/supportdata-N5KNU-LB5-QTH-2-ESN-2025-3-8-10-58.tar.zip)

aanon4, 2025-03-08:
The support data above is from just after a reboot, after the hotspot is disabled. I'm guessing that the hotspot, which requires both hostapd and wpa_supplicant to run, is just too much when also running the new babel stuff. For 64M that's annoying, but let's see if this stabilizes things first before thinking on next steps.

kp4djt, 2025-03-08:
OK, just got back in and found it linked to the entry point node, the link
numbers were very good and I saw
that the hotspot was off, It does certainly make a difference. I am turning
those off on all of them. We turn
them on to do link alignment but should turn them off after doing so I
guess.

KG7GDB, 2025-03-09:
Ok, the same bug happened to me today as well. It has happened several times over the past week where I cannot load the localnode.local.mesh status page.
I have an hAP ac lite running 20250307-8b861663 nightly.
The node had been unused all day and I attempted to login via WiFi. I always use the WiFi hotspot (5 GHz channel 165) when using the hAP ac lite. It is main reason to use a hAP so you can use the mesh from various devices.  As above, I was able to navigate to other mesh sites I had bookmarked and use the local WAN. 
I was able to SSH into the hAP ac lite and save the support file before rebooting.
In case people need to generate a support file from the CLI, in Linux you run the command:
/usr/local/bin/supportdata 
which generates a support file in the /tmp directory, then you have to scp it to your desktop. You exit SSH then type:
scp -O -P2222 root@nodename:/tmp/supportdata.tar.gz ~/Desktop/

[supportdata.tar.gz](https://github.com/user-attachments/files/19147740/supportdata.tar.gz)

I also attempted to restart the webpage with the command /etc/init.d/uhttpd restart. I ran the free command before and I will attach the terminal screenshot:

[image]

This worked briefly and the status page loaded i Firefox, but I clicked on it and it went gray and unresponsive. I could see the load levels were 6.43, 3.02, 1.16, then I went back to the terminal which was also unresponsive. At this point, the node rebooted itself as I have the hardware watchdog.

Now the hAP is running again and the load levels are 0.08  0.05  0.09 I have free flash at 8.888 and free ram at 9.064 but this varies every time I click the page. 
I sure wish we could fix this! I need that hotspot more than I need Babel. -Brett, KG7GDB

aanon4, 2025-03-09:
So it's not a memory leaks as best we can tell - it's legitimately running out of memory. Disabling the hotspot gives you back some because less daemons are running. Tomorrow we'll switch out the offending daemon to see if that reduces its footprint as there's some information indicating an interaction between the 6.X kernels and hostapd can cause excess memory consumption.

kp4djt, 2025-03-09:
I too have observed the same issue with the hAP ac lite. Indeed, the one
that was at the far end of the link
where we were trying to get the data from initially had this issue but the
last nightly seemed to calm it down
as I have had no issues with it.

I also have a hap ac lite in my van as my mobile node, it has the same
issue, this afternoon I went to check out
a repeater site that is also a node site, I went to access the node in the
van but was unable to obtain the web
page. The node was there, the web server was down. I had to reboot the node
then I could get to it.

To date the devices that I have seen this issue on are the hap ac lite and
the litebeam series of radios.