aredn/aredn #1964: New interface doesn't accept port forwarding ranges

AI7NC, 2025-03-08:
### Describe the bug
A clear and concise description of what the bug is.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
A clear and concise description of what you expected to happen.

### Support data
Please add the support data for all nodes mentioned in this bug report (**Required**)

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-03-08:
You don't specify the range twice, only in the first box. The mapping is automatic.

AI7NC, 2025-03-09:
Ok, I can confirm that it does work if you include the start of the range in the local port field and leave the end of the range out. 

[image]

closed 2025-03-09 by aanon4