aredn/aredn #1966: AR300M fails to upload firmware

hwdornbush, 2025-03-09:
I have tried two AR300M16 devices trying to upload the AREDN firmware using the System > Upgrade > Local Upgrade.  I tried 3 versions of the .bin file, 3.23, 3.24, and 3.25.  In all cases, the upload is shown as successful, but the Firmware Verification fails.  I checked the SHA256 and it matches what is shown when I download the firmware from thee AREDN site.

[image]

aanon4, 2025-03-09:
There are different variants of the AR300M out there. Is there a screenshot you can post with the details of this specific one? Thanks.

hwdornbush, 2025-03-09:
This was from a GL-AR300M16.  I also tried it with a GL-AR300M16-ext with the same results.

[image]

hwdornbush, 2025-03-09:
When I powered up the AR300M today, it offered a newer version of the firmware, V4.3.22 which gave me the same result.  I downloaded an earlier version of the firmware, V4.3.7 and it also gave me the same result of "Testing the firmware with sysupgrade failed."

k1ky, 2025-03-10:
Basically same result with HAP ac Lite - have not been successful installing 0310 nightly from 0309.

aanon4, 2025-03-10:
@hwdornbush There are in fact three different builds for devices all called AR300M and right now we only build two of them. I will add the third, which is the one ending -NOR. I wish they had used slightly different model numbers.

aanon4, 2025-03-10:
@hwdornbush Is there an internal UI page which shows the exact details of the hardware? I would like to get more internal details which are not given on the external label.

hwdornbush, 2025-03-10:
Tim,
I am not aware of an internal page other than
https://gl-inet.com/products/gl-ar300m/

This page shows that the models like GL-AR300M have 128mb nand flash while
those like GL-AR300M16 do not, but only have 16mb nor flash.

When I  first tried to flash AREDN on a AR300M16 a few years ago, it worked
fine while that is no longer true.

Bill

aanon4, 2025-03-10:
@hwdornbush I miscounted earlier - there are 4 hardware versions all with the same name. Unfortunately the web page link is just going to describe the generic product and not the specific instance you have in-hand. Hopefully you can find some UI on the device itself which describe the exact hardware specifics.

hwdornbush, 2025-03-11:
I understand there are multiple versions. I downloaded the version that matches my device, AR300M16.  But it will not load onto a new AR300M16 device.  Nor will it load on another AR300M16 device I have had for a few years.  The firmware from GL-inet will load so I suspect something is different in the AREDN firmware from the prior firmware that did work, about a year ago.  So how do I get a good version of the AREDN firmware that will load on an AR300M16 device?

tmitchell01, 2025-03-13:
I just purchased the same model listed above. I am including a screen shot of the system information if this helps supply any additional information. Will see what other details are visible to supply here.

[image]

hwdornbush, 2025-03-13:
There are two models of this device: the AR300M and the AR300M16.  I have the problem with the AR300M16.

kc2cbd, 2025-03-29:
Technically there are only 2 versions with 2 variants of each. The AR300M is the nand device and comes with and without external antennas. AR300M16 is the nor device and comes with and without external antennas. There was also a GL-AR300M-Lite which is discontinued. It only came with 1 ethernet port and no USB and was a nor device.

https://www.gl-inet.com/products/gl-ar300m/?utm_source=website&utm_medium=menubar

tmitchell01, 2025-05-03:
Does someone need the GL-AR300M to troubleshoot this issue?
More than happy to ship to whoever would be able to assist on getting this work.