aredn/aredn #1969: Some duplicate radio info shown on 5 GHz nodes

Orv, 2025-03-11:
### Describe the bug
The Radio window duplicates some of the 5 GHz radio information at the bottom of the window

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes

### Expected behavior
A clear and concise description of what you expected to happen.
Radio data should not be duplicated

### Support data

[supportdata-KJ6TTN-PointedToSimiNorth-5G-2025-3-11-20-46.tar.gz](https://github.com/user-attachments/files/19197822/supportdata-KJ6TTN-PointedToSimiNorth-5G-2025-3-11-20-46.tar.gz)

### Screenshots

[image]

### Additional context
Shows on several different 5 GHz nodes; does NOT show on 2 GHz TP-Link.
This issue does not occur on a 5 GHZ node running nightly build 20250304.

aanon4, 2025-03-11:
https://github.com/aredn/aredn/pull/1971

Orv, 2025-03-12:
Latest NB.  Duplicate 5 GHz info on this 5 GHz Nanobeam was replaced with phantom 2.4 GHz info.

[image]

[supportdata-W6BI-QTH-5G-2025-3-12-11-4.tar.gz](https://github.com/user-attachments/files/19215211/supportdata-W6BI-QTH-5G-2025-3-12-11-4.tar.gz)

aanon4, 2025-03-12:
Phantom? A nanobeam has a 2.4GHz radio.

jwheeler188, 2025-03-12:
> [@jwheeler188](https://github.com/jwheeler188) Is this on today's build?

That was on FW 20250310-c57674d3 I didn't realize a new version was released this morning. I will get it loaded and will test when I get home from the office. I should also have a hAP3 and a litebeam-5-ac-gen2 waiting for me on the porch when I get home. I might try putting the latest nightly on the hAP3 tonight if time permits.

Orv, 2025-03-12:
Dual-band Nanobeam - I never knew that.  Apparently you can't use them simultaneously (could that be a thing? Would really be handy)  After switching the radio to 2.4 GHz and configuring it, it was unable to see the hilltop 2.4 GHz node 1.5 miles away.  I'll keep investigating.

aanon4, 2025-03-12:
The 2.4 radio is a management radio - not attached to the main antenna but works for local access.

Orv, 2025-03-12:
Then it's working as expected, I guess.

closed 2025-03-12 by Orv

jwheeler188, 2025-03-13:
> [@jwheeler188](https://github.com/jwheeler188) Is this on today's build?

Ok, I got home and I am still having the same issue on the new build, but only on my hAP lite. The hAP3 is working fine.

aanon4, 2025-03-13:
This looks to be the final part of this - https://github.com/aredn/aredn/pull/1978

jwheeler188, 2025-03-13:
> This looks to be the final part of this - [#1978](https://github.com/aredn/aredn/pull/1978)

This fixed the problem!!