aredn/aredn #1970: 3.24.10  New UI stylesheet not loading when localnode is pre 3.26.6  Text aligns on left side of screen.

jskier56, 2025-03-11:
### Describe the bug
When  localnode.local.mesh is on a release, 3.22.12, that does not support the 3.24.10 new UI the stylesheet is not loaded.  This results in the text aligning on the left side.  This makes using the interface a challenge  unless the old UI is available.   Later releases that don't support old UI are not usable. 

- [ ] Does this issue still occur on the latest version? (**Required**)
This issue is present in releases up to nightly build 20250310 .

### Expected behavior
The NEW GUI displays with data in panels.   A change to the stylesheet path that used the FQDN or IP prefix should enable the stylesheet to be found. 

### Support data
We've determined the stylesheet has a path prefix of  localnode.local.mesh .   If the local node is on an older release the style sheet is not found so it does not load and the  screen paints with all text left aligned.   

### Work around 
If practical upgrade the  localnode to a release that supports the new UI.  

If you cannot upgrade the loccal node   Define localnode,  localnode.local.mesh  in the  system /etc/hosts file using an IP address of a device that supports the new UI.    I.E   10.156.189.216 localnode localnode.local.mesh  K5SKR-QTH

aanon4, 2025-03-11:
Please provide screenshots - thanks.

jskier56, 2025-03-11:
Have to down grade nodes.

73s

John K5SKR

jskier56, 2025-03-11:
Original information updated with screen shots

aanon4, 2025-03-12:
The following should happen:

1. The node will attempt to pull the relevant stylesheets from your localnode, regardless of what node you're visiting.
2. If that fails, if should pull the stylesheet from the node your visiting.

So I'm assuming (1) is failing but somehow not triggering (2). It would be useful to see the result of the following command:
```
curl -v http://localnode.local.mesh/a/css/user.de4dddabdbe70abebcf6e2d33ccc1858.css
```

aanon4, 2025-04-12:
No feedback - closing

closed 2025-04-12 by aanon4