aredn/aredn #1975: Local nodes not shown

kp4djt, 2025-03-12:
### Describe the bug
After installing nightly 20250312-7805d787 local nodes no longer shown

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
This particular node should have 

### Support data

[supportdata-KP4DJT-hAP-home-2025-3-12-15-23.tar.gz](https://github.com/user-attachments/files/19217322/supportdata-KP4DJT-hAP-home-2025-3-12-15-23.tar.gz)

### Screenshots
Shot of local node area, is empty
[image]
Shot of neighbor local node showing all of DtD devices

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2025-03-12:
Could you let me know which firmware version each of the neighbors is running? Thanks.

aanon4, 2025-03-12:
It's probably this though - https://github.com/aredn/aredn/pull/1976

kp4djt, 2025-03-12:
TP-Link CPE210 v3
20250306-620f515b
MikroTik RouterBOARD LHG 5nD                         20250301-0bef22a0
MikroTik LHG 5 ac (RBLHGG-5acD)                     20250306-620f515b
MikroTik RouterBOARD 921GS-5HPacD-15s       20250301-0bef22a0

That should be it.

kp4djt, 2025-03-13:
Appears that issue is with the ac3, I just  updated an ac2 for grins and it
looks great:[image: image.png]

closed 2025-03-13 by aanon4

kp4djt, 2025-03-17:
Issue was  DtD switch  not powered, node could ot see other nodes. Not sure if this happened at time of upgrade, which was done remotely or it the original upgrade did have this issue but the switch failure continued to imitate it.

The system is now working as it should.