aredn/aredn #1982: WAN Access into AREDN device not working

jwheeler188, 2025-03-14:
### Describe the bug
I have no remote WAN access at all to any of my device. Even with all sliders green for web/telnet/ssh access. It is deceiving, because when plugged directly to the back of the hAP's, I can ping the WAN address, but technically that traffic is originating from the LAN side. If I connect directly to my home router, I have no access across my home network into the WAN port of the hAP.

Case 1: hAP connected to WAN port 1 or WAN Client hAP WAN IP address of 10.0.0.64. Computer connected directly to HAP. Computer has IP address of 10.178.72.37.

Test 1: Target host address for testing of 10.0.0.64: Browser access, SSH, ping, telnet all successful.

Case 2: hAP connected to WAN port 1 or WAN Client hAP WAN IP address of 10.0.0.64. Computer connected directly to home router. Computer has IP address of 10.0.0.188.

Test 2: Target host address for testing of 10.0.0.64: Browser access, SSH, ping, telnet all fail.

In either case, node does have working internet. 

If I connect my computer directly to a lan port on the hAP, my computer will have full internet and mesh access. Additionally, when pooling the FW and software lists from the UI, the package names will populate, and I can download software directly from the GUI.

- [ ] Does this issue still occur on the latest version? (**Required**)
YES 

### Expected behavior
With WAN access enabled for Telnet, SSH, and Web in the last stable release, I could access the node as expected across my home network when not directly connected to the switch. 

### Support data
Please add the support data for all nodes in this bug report (**Required**)

[supportdata-AA5JW-Test-Node-2025-3-14-18-33.tar.gz](https://github.com/user-attachments/files/19252909/supportdata-AA5JW-Test-Node-2025-3-14-18-33.tar.gz)

### Screenshots

[image]

[image]

[image]

If applicable, add screenshots to help explain your problem.

### Additional context
N/A

aanon4, 2025-03-14:
AREDN uses 10.0.0.0/8 for its network.  Having a WAN address also in this network isn't going to go well. You might get lucky if the specific 10.0.0.0/24 doesn't overlap with anything on your local areas mesh and I've seen it work; but I've mostly seen it not work. When I started with AREDN my home used 10.0.0.0/24 and I literally changed everything to avoid this problem.

jwheeler188, 2025-03-15:
Re-designing and re-deploying my home network has resolved the issue. Thanks!!

closed 2025-03-15 by jwheeler188

K5RA, 2025-03-15:
Cleaning up devices and port forwarding in my ISP's router appears to have solved my issues (Bug #1945).  Thanks to all for inputs which have assisted getting this fixed.